amzn/amzn-drivers #1: Please submit this for inclusion in the Linux kernel

bootc, 2016-06-28:
Maintaining an out of tree driver is unhelpful to you, unhelpful to your users and unhelpful to Linux kernel developers. Having this included in the kernel means everyone can benefit from using it on AWS without having to deliberately go and jump through hoops to install it. Please, we'll all love you more for doing so!

mswilson, 2016-06-28:
Hi Chris,

The driver was initially submitted on June 13th. See http://thread.gmane.org/gmane.linux.network/416446

It was also posted a while back while in development: http://thread.gmane.org/gmane.linux.network/404282

A v2 in response to reviews is being prepared.

Kind regards,

Matt

closed 2016-06-28 by mswilson

netanelbelgazal, 2018-11-27:
> > A v2 in response to reviews is being prepared.
> 
> Was any progress towards including this in the Linux kernel ever made? It seems like a big hurdle to have to recompile my Linux kernel every time there's a new version of AWS ENA drivers released. I'm using Debian, if there's some kind of `sudo apt-get install amzn-drivers-ena` I can do, I would also like that.

Yes,
We constantly sync the ENA driver to the Linux kernel. (ENAv2 was already merged and it is expected to be part of kernel 4.20).
Beside that, you don't have to recompile the entire kernel every time there is a new release.
You can compile the driver as a module and use DKMS for automatic compilation.

bootc, 2018-11-27:
There isn't a DKMS package for ENA in Debian, sadly, and by the time buster is released I would expect it to mostly be unnecessary. ENA 1.0.2 is in Debian's stable kernels, so old but still generally usable, and ENA 1.5.0K is in the current 4.18 kernel in buster (but that will almost certainly change by the time buster is released).

To be fair Amazon deserve plenty of credit for upstreaming this driver in the way they have.

bootc, 2018-11-27:
I should add if Amazon don't think the 1.0.2 driver in Debian is adequate then they should reach out to the appropriate people in Debian to get it updated. I'm not the right person, but I can probably find the right person/people to have that discussion with if necessary.

netanelbelgazal, 2018-11-27:
If you want to use DKMS to build the ENA driver you can follow the instructions propose by @akiyano 
https://github.com/amzn/amzn-drivers/issues/88

@bootc 
We recommend upgrading the ENA driver from 1.0.2.
We'll appreciate if you can connect us to the right people inside Debian.

nmeyerhans, 2018-11-30:
@bootc, @netanelbelgazal I've opened a [merge request](https://salsa.debian.org/kernel-team/linux/merge_requests/77) to get the 2.0.2 driver backported to the linux 4.19 kernel that should be included in buster. It was a pretty straightforward change. Please feel free to leave feedback there if you have access, or here otherwise.

I hope to put something similar together for stretch, but with its 4.9 kernel it's going to be a bit more of a challenge.