amzn/amzn-drivers #3: Module Verification Failure on Ubuntu 16.04 AMI

sheyd, 2016-09-12:
When following instructions for Ubuntu 16.04 (and using systemd dynamic network port disabling like required in CentOS/RHEL 7), following error from syslog:

`ena: module verification failed: signature and/or required key missing - tainting kernel`

My knowledge of module kernel signing is limited. Happy to dive deeper into the code or run further de-bugging if you can point me in the right direction.

sheyd, 2016-09-12:
Specific AMI: `ami-2ef48339` in US-East-1 Region.

netanelbelgazal, 2016-09-12:
The message you get refers to kernel module signature verification: an
optional feature that requires loadable modules to be signed by the linux
distribution vendor.

The ENA kernel module driver you built is considered “built out of tree”
and would not have kernel signing key which is exactly the message you
receive.

This message is considered a warning and does not break any functionality
of the driver unless the kernel have the parameter CONFIG_MODULE_SIG_FORCE
on.

I checked the AMI you gave us and I believe the parameter is not set.

Going forward, I will work on finding a way to avoid this, or update the
release note to consider this

Please let me know If above feedback helps, and if I can close the issue

sheyd, 2016-09-13:
@netanelbelgazal Thank you for the prompt response. The feedback does indeed help. The I can confirm the module does load on boot. The only thing is the interface of `eth0` still uses the ixgbevf driver (run using `ethtool -i eth0`). Is there a step missing here? Here is simple bash script used to run all the steps in order:

netanelbelgazal, 2016-09-13:
Which type of instance are you trying to launch?
Currently ENA is only available on X1 instances and on those instances does not have ixgbevf devices.

For more information please refer to:
https://aws.amazon.com/about-aws/whats-new/2016/06/introducing-elastic-network-adapter-ena-the-next-generation-network-interface-for-ec2-instances/

sheyd, 2016-09-13:
Wow, total oversite on my end. I was using an instance in the M4 series. Sorry to cause the run-around @netanelbelgazal. Please feel free to close ticket.

closed 2016-09-14 by sheyd