amzn/amzn-drivers #5: Compilation fails on kernel 3.10.0-514.2.2.el7.x86_64

egriffith, 2016-12-13:
DKMS install, on a CentOS 7 / RHEL 7 instance, fails with DKMS errors:

DKMS make.log for amzn-drivers-1.1.2 for kernel 3.10.0-514.2.2.el7.x86_64 (x86_64)
Tue Dec 13 16:56:11 UTC 2016
make: Entering directory `/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena'
make -C /lib/modules/3.10.0-514.2.2.el7.x86_64/build M=/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-514.2.2.el7.x86_64'
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
  CC [M]  /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.h:43:0,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/kcompat.h:298:0: warning: "GENMASK" redefined [enabled by default]
 #define GENMASK(h, l) (((U32_C(1) << ((h) - (l) + 1)) - 1) << (l))
 ^
In file included from include/linux/kernel.h:10:0,
                 from include/linux/cpumask.h:9,
                 from include/linux/cpu_rmap.h:13,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:36:
include/linux/bitops.h:21:0: note: this is the location of the previous definition
 #define GENMASK(h, l) \
 ^
In file included from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.h:43:0,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/kcompat.h:299:0: warning: "GENMASK_ULL" redefined [enabled by default]
 #define GENMASK_ULL(h, l) (((U64_C(1) << ((h) - (l) + 1)) - 1) << (l))
 ^
In file included from include/linux/kernel.h:10:0,
                 from include/linux/cpumask.h:9,
                 from include/linux/cpu_rmap.h:13,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:36:
include/linux/bitops.h:24:0: note: this is the location of the previous definition
 #define GENMASK_ULL(h, l) \
 ^
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_select_queue’:
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:2203:3: error: implicit declaration of function ‘__netdev_pick_tx’ [-Werror=implicit-function-declaration]
   qid = __netdev_pick_tx(dev, skb);
   ^
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-514.2.2.el7.x86_64'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena'

zorikm, 2016-12-13:
Thanks for catching this!
We are working to resolve these issues.

Zorik

zorikm, 2016-12-17:
HI Eric,
We'll upload a fix next week, after our testing is done. In the meantime I'm attaching a patch that should unblock you. 
Thanks,
Zorik
[0099081480_1481751768_rhel_7_3.txt](https://github.com/amzn/amzn-drivers/files/658541/0099081480_1481751768_rhel_7_3.txt)

egriffith, 2016-12-19:
Thank you Zorik!

-- ericgri@

rsmoorthy, 2016-12-27:
Exactly the same problem.

The patch has been helpful.

netanelbelgazal, 2017-01-03:
The compilation was fixed by driver version 1.1.3 (3ac3e0b)
Let me know if you have further problems.

closed 2017-01-03 by netanelbelgazal