amzn/amzn-drivers #6: Fails to compile on Centos 6

jsharpe, 2016-12-20:
There are multiple issues with redefinitions in kcompat.h when compiling against the latest centos 6 kernel.

zorikm, 2016-12-21:
Hi jsharpe.
Can you please specify the exact CentOs version and kernel version you are using.

mlnkrlc, 2016-12-21:
@zorikm 
Here's a build log from 2.6.32-642.6.2.el6.x86_64 that comes by default with RHEL AMI ami-b990d8ca (eu-west-1):

`DKMS make.log for amzn-drivers-1.1.2 for kernel 2.6.32-642.6.2.el6.x86_64 (x86_64)
Wed Dec 21 05:50:03 EST 2016
make: Entering directory `/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena'
make -C /lib/modules/2.6.32-642.6.2.el6.x86_64/build M=/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/2.6.32-642.6.2.el6.x86_64'
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
  CC [M]  /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.h:43,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/kcompat.h:47:34: error: generated/utsrelease.h: No such file or directory
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/kcompat.h:228:1: warning: "EPROBE_DEFER" redefined
In file included from include/linux/notifier.h:12,
                 from include/linux/memory_hotplug.h:6,
                 from include/linux/mmzone.h:717,
                 from include/linux/gfp.h:4,
                 from include/linux/cpu_rmap.h:14,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:36:
include/linux/errno.h:19:1: warning: this is the location of the previous definition
In file included from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.h:43,
                 from /var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/kcompat.h:234: error: redefinition of ‘ether_addr_equal’
compilation terminated due to -Wfatal-errors.
make[2]: *** [/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/2.6.32-642.6.2.el6.x86_64'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/1.1.2/build/kernel/linux/ena'`

zorikm, 2016-12-21:
Thanks @mlnkrlc,
The current driver version supports kernel versions starting with 3.2. That's why you see so many compilation errors. Our apologies for not mentioning this in Release Notes, we'll update them.
We are working these days on backporting ENA driver to 2.6.32. We are planning to upload a patch next week.

netanelbelgazal, 2017-01-03:
@mlnkrlc 
I fixed CentOS 6 build by driver version 1.1.3 (3ac3e0b)
Let me know if you still have problems

closed 2017-01-03 by netanelbelgazal