amzn/amzn-drivers #7: kabi compliance ?

alex-miro, 2016-12-24:
AMZN recommends dkms as a method for keeping up with kernel updates.
This approach does not always work for enterprise customers for policy reasons.
A quick check against RHEL7.3 whitelist shows a lot of non compliant symbols.
Any plans to make it kabi  whitelist compliant - at least for RHEL  ?

netanelbelgazal, 2016-12-24:
We are going to release a fix for RHEL 7.3 compilation in a week.
For the upcoming release we won't have kobi support.

I noted your request and we'll add kabi whitelist compliant for later releases.

zorikm, 2017-05-18:
Hi @alex-miro,
Thank you for your question.
At this point we decided not to maintain kABI compliancy, as it might imply significant code complications. 
We work together with our RHEL/CentOS partners to make sure all potential issues are resolved by the time they release their kernel updates.

closed 2017-05-18 by zorikm