amzn/amzn-drivers #8: Feature request: XDP support

c4milo, 2017-01-05:
are there any plans to support XDP? 

https://www.iovisor.org/technology/xdp

zorikm, 2017-01-06:
Hi Camilo,
We do have plans to support XDP in future releases.
Still do not have exact dates.
Having said that, we definitely encourage community to contribute.

tgraf, 2017-09-22:
> Having said that, we definitely encourage community to contribute.

How we can help? We are developing https://github.com/cilium/cilium and would love to contribute.

borkmann, 2018-03-14:
@zorikm @netanelbelgazal Do you have any new updates or rough ETA by now for integrating XDP on ena?

sbernard31, 2019-05-21:
Any news about **XDP** support ? 

Does it exist any alternative/workaround to be able to use **XDP** in AWS ? (maybe using `virtio-net` driver?)

nafeabshara, 2019-05-21:
we actively working on this, top of priority stack

dwmw2, 2019-05-22:
net-next just opened for submissions....

gtzalik, 2020-01-19:
XDP support was added to v2.2 and to net-next.
Closing

closed 2020-01-19 by gtzalik