amzn/amzn-drivers #9: iflib support

kev009, 2017-02-20:
Have a look at the iflib man pages on -CURRENT: iflibdd(9), iflibtxrx(9)

The driver should be using this for queue management and txrx.  I don't see anything in the NIC design that is mutually exclusive to this.

kev009, 2017-02-20:
cc @mattmacy

gtzalik, 2017-02-27:
iflib is a new feature and we currently checking the implications of adding this support

kev009, 2017-02-27:
@gtzalik thanks for checking.  We will be MFCing everything to 11-STABLE and 10-STABLE in cooperation with Intel.  @mattmacy has been working on small packet and forwarding performance in https://github.com/mattmacy/networking/commits/HEAD_MERGE/iflib-ixgbe-current, any concerns you have for ENA can be brought to his attention.

alexissavin, 2019-07-25:
Support for iflib would be greatly appreciated as this would also permit to use NetMap on ENA driver in the future.