amzn/amzn-drivers #10: Error when compiling on RHEL 6.7

ambud, 2017-03-23:
Compilation fails on RHEL 6.7 (EC2), headers and devel packages are installed

make -C /lib/modules/2.6.32-696.el6.x86_64/build M=/home/ec2-user/amzn-drivers/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/2.6.32-696.el6.x86_64'
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_netdev.o
/home/ec2-user/amzn-drivers/kernel/linux/ena/ena_netdev.c: In function ‘ena_intr_msix_io’:
/home/ec2-user/amzn-drivers/kernel/linux/ena/ena_netdev.c:1264: error: implicit declaration of function ‘__napi_schedule_irqoff’
compilation terminated due to -Wfatal-errors.
make[2]: *** [/home/ec2-user/amzn-drivers/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/home/ec2-user/amzn-drivers/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/2.6.32-696.el6.x86_64'
make: *** [all] Error 2

zorikm, 2017-03-23:
Hi ambud.
Can you please specify what AMI ID (ami-xxxxxxxx) are you using?

ambud, 2017-03-23:
Hi @zorikm, it's a private AMI running RHEL 6.7

zorikm, 2017-03-24:
It seems that you are using a kernel version that belongs to RHEL 6.9, so let's first check the RHEL version. We need to know it in order to help you to resolve this issue.
 What's the output of:
>cat /etc/redhat-release

ambud, 2017-03-24:
Thanks for the quick response and the pointer about kernel version @zorikm 

```cat /etc/redhat-release  >> Red Hat Enterprise Linux Server release 6.7 (Santiago)```

I double checked the version of kernel-devel that is installed based on your RHEL 6.9 comment above. 

Looks like what happened was ```yum update``` installs the latest kernel and ```yum install kernel-devel``` installs the latest **kernel-devel** package. This caused the kernel to be upgraded causing issues with the driver build.

Thank you for the pointer, I would like to keep this issue open for a little bit longer until we have it deployed in our private AMI.

zorikm, 2017-03-24:
Thanks @ambud,
Basically we would recommend to use:
yum install kernel-devel-$(uname -r)
This would ensure that you get kernel-devel for the kernel you currently have.
We'll update our README.
Please let us know if this resolves your issue.

jstrunk, 2017-04-11:
I had a similar issue getting this working with Centos 6. Something changed in the RHEL 6 kernel between `2.6.32_573.el6.x86_64` and `2.6.32-696.el6.x86_64` that breaks compatibility.

Centos 6 launched with `ami-c299c2a7` in us-east-2 boots with `2.6.32_573.el6.x86_64`, but kernel-{devel,headers} packages are unavailable in their repos for that version. After running `yum update` to get to `2.6.32-696.el6.x86_64` I see the same error message shown in the original issue.

I confirmed that RHEL 6 as booted from `ami-274ba847` in us-west-2 can build ena as the kernel-{devel,headers} packages exist for `2.6.32_573.el6.x86_64`. After running `yum update` to get to `2.6.32-696.el6.x86_64`, I get the same error. Also, the ena module built earlier no longer runs on `2.6.32-696.el6.x86_64`.

yastreb78, 2017-04-12:
Thanks for report @jstrunk ,
We are going to update github with the fix soon , meanwhile you can use this patch:

diff --git a/kernel/linux/ena/kcompat.h b/kernel/linux/ena/kcompat.h
index cea96dc..c47b0b3 100644
--- a/kernel/linux/ena/kcompat.h
+++ b/kernel/linux/ena/kcompat.h
@@ -464,8 +464,7 @@ static inline void netdev_rss_key_fill(void *buffer, size_t len)
 #endif
 
 #if ( LINUX_VERSION_CODE < KERNEL_VERSION(3,19,0) ) && \
-    !(SLE_VERSION_CODE && SLE_VERSION_CODE >= SLE_VERSION(12,0,0)) && \
-    !(RHEL_RELEASE_CODE && RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(6,9))
+    !(SLE_VERSION_CODE && SLE_VERSION_CODE >= SLE_VERSION(12,0,0))
 
 static inline void napi_schedule_irqoff(struct napi_struct *n)
 {

ambud, 2017-04-30:
Verified on our AMI, closing

closed 2017-04-30 by ambud