amzn/amzn-drivers #11: napi_hash_add is now static

mlosapio, 2017-04-25:
Compilation error against 4.10 kernel

```
/root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena/ena_netdev.c:1493:3: error: implicit declaration of function 'napi_hash_add' [-Werror=implicit-function-declaration]
   napi_hash_add(&adapter->ena_napi[i].napi);
   ^
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
```

Possibly related to this: https://www.mail-archive.com/netdev@vger.kernel.org/msg136423.html

mlosapio, 2017-04-25:
https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ena_netdev.c#L1493

zorikm, 2017-04-25:
Thanks Michael for raising this.
We'll look into it and get back to you asap.

closed 2017-05-14 by netanelbelgazal