amzn/amzn-drivers #14: `LL_FLUSH_FAILED` and `LL_FLUSH_BUSY` moved in 4.11

drcrallen, 2017-05-05:
http://elixir.free-electrons.com/linux/v4.11/source/include/net/busy_poll.h

zorikm, 2017-05-06:
Thanks, we are in the process of updating and testing the driver for 4.11. The patch will be available soon. Please stay tuned.

kwilczynski, 2017-05-08:
@drcrallen the kernel tree driver diverged slightly :(

These are other things which (potentially) needed to be fixed:

- https://git.kernel.org/pub/scm/linux/kernel/git/mcgrof/linux-next.git/commit/?id=37fabbf4d489cc2e1cbf7cde816d9453a65ddfb7
- https://git.kernel.org/pub/scm/linux/kernel/git/mcgrof/linux-next.git/commit/?id=93411c27d6504933d2d81399a32bb52e24b06839
- https://git.kernel.org/pub/scm/linux/kernel/git/mcgrof/linux-next.git/commit/?id=4244de1c64ded7f5438717bdce3fa074efd20efb
- https://git.kernel.org/pub/scm/linux/kernel/git/mcgrof/linux-next.git/commit/?id=bc1f44709cf27fb2a5766cadafe7e2ad5e9cb221

I have fixed most of the issues already by myself, to be able to build ENA against 4.10 and newer (especially, since 4.11 branch was recently cut), and after reaching out to @netanelbelgazal he suggested that it would be prudent to introduce updates so that they are handled via the `kcompat.h`.

netanelbelgazal, 2017-07-18:
Issue fixed

closed 2017-07-18 by netanelbelgazal