amzn/amzn-drivers #15: memory corruption panic on FreeBSD11.0 with ena0

kyenduri, 2017-05-09:
I hit a panic with ena driver on FreeBSD11.0. Built the ena driver for FreeBSD11.0 and tried it on a AWS instance (r4.large). The system comes up and then panics. The console messages look -
...
ena0: <ENA adapter> mem 0x83000000-0x83003fff at device 3.0 on pci0
ena0: Elastic Network Adapter (ENA)ena v0.6 (Jan 12, 2017)

ena0: Allocated msix_entries, vectors (cnt: 3)
ena0: initalize 2 io queues
ena0: Ethernet address: 02:b8:a4:5f:b0:a2
...
ena0: link state changed to UP
...
ena0: ena_setup_io_intr vector: 2
ena0: ena_setup_io_intr vector: 3
ena0: queue 0 - cpu 0
ena0: queue 1 - cpu 1
ena0: RX Soft LRO[0] Initialized
ena0: RX Soft LRO[1] Initialized
DHCPDISCOVER on ena0 to 255.255.255.255 port 67 interval 5
DHCPOFFER from 172.31.16.1
DHCPREQUEST on ena0 to 255.255.255.255 port 67
DHCPACK from 172.31.16.1
ena0: device is going DOWN
ena0: ena_free_rx_resources qid 0
ena0: ena_free_rx_resources qid 1
ena0: ena_setup_io_intr vector: 2
ena0: ena_setup_io_intr vector: 3
ena0: queue 0 - cpu 0
ena0: queue 1 - cpu 1
ena0: RX Soft LRO[0] Initialized
ena0: RX Soft LRO[1] Initialized
bound to 172.31.29.238 -- renewal in 1800 seconds.
Starting Network: lo0 ena0.
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> metric 0 mtu 16384
        options=600003<RXCSUM,TXCSUM,RXCSUM_IPV6,TXCSUM_IPV6>
        inet 127.0.0.1 netmask 0xff000000
        groups: lo
ena0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 9001
        options=1043a<TXCSUM,VLAN_MTU,VLAN_HWTAGGING,JUMBO_MTU,LRO,VLAN_HWFILTER>
        ether 02:b8:a4:5f:b0:a2
        inet 172.31.29.238 netmask 0xfffff000 broadcast 172.31.31.255
        media: Ethernet autoselect (10Gbase-T <full-duplex>)
        status: active
Enabling pfpanic: Memory modified after free 0xffff8480059bf150(8) val=0 @ 0xffff8480059bf150

cpuid = 0
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xffff860039949140
vpanic() at vpanic+0x114/frame 0xffff860039949180
panic() at panic+0x43/frame 0xffff8600399491e0
trash_ctor() at trash_ctor+0x61/frame 0xffff860039949200
uma_zalloc_arg() at uma_zalloc_arg+0x588/frame 0xffff860039949270
counter_u64_alloc() at counter_u64_alloc+0x1b/frame 0xffff860039949290
pfioctl() at pfioctl+0x16b5/frame 0xffff860039949780
devfs_ioctl_f() at devfs_ioctl_f+0x159/frame 0xffff8600399497e0
kern_ioctl() at kern_ioctl+0x15e/frame 0xffff860039949840
sys_ioctl() at sys_ioctl+0x173/frame 0xffff860039949920
amd64_syscall() at amd64_syscall+0x305/frame 0xffff860039949ab0
Xfast_syscall() at Xfast_syscall+0xfb/frame 0xffff860039949ab0
--- syscall (54, FreeBSD ELF64, sys_ioctl), rip = 0x800dfc3ca, rsp = 0x7fffffffce18, rbp = 0x7fffffffda40 ---

One thing that looks odd is that the link goes DOWN and then comes UP during DHCP.

yastreb78, 2017-05-09:
kyenduri@ Thanks your report.
Would it be possible to share your AMI with us? (AWS ID 890902708306 & 255982136585)

Also,we will be soon publishing new version 0.7, which should be fixing few stability issues.

kyenduri, 2017-05-09:
Unfortunately, this AMI can not be shared. It is closest to the public FreeBSD AMI ami-0307a163. I will retry with version 0.7 once it is out.

gtzalik, 2017-05-11:
kyenduri,  is it possible for you to share the following files to assist us in reproducing the issue
/etc/hosts, /etc/pf.conf, /etc/pf.os?
If possible, feel free to send it to me by email

kyenduri, 2017-05-12:
I got a fix for this. The driver is doing ena_free_counters() twice for rx_stats. I no longer see the panic with this patch - 
--- ./freebsd/sys/dev/ena/ena.c	2017-05-05 13:49:22.000000000 -0700
+++ ena.c	2017-05-11 18:10:44.000000000 -0700
@@ -879,10 +879,6 @@
 	ENA_MEM_FREE(adapter->ena_dev->dmadev, rx_ring->rx_buffer_info);
 	rx_ring->rx_buffer_info = NULL;
 
-	/* free counters... */
-	ena_free_counters((counter_u64_t *)&rx_ring->rx_stats,
-	    sizeof(rx_ring->rx_stats));
-
 	/* ... and destroy mutex. Voila! */
 	mtx_destroy(&rx_ring->ring_mtx);

gtzalik, 2017-05-21:
thanks @kyenduri
we already fix the above issue in our latest code.
I will upload new version with the fix in few days

gtzalik, 2017-06-07:
the issue fixed on f171fc9bd2189156593db533f00b4c14c58e3bbb

closed 2017-06-07 by gtzalik