amzn/amzn-drivers #16: FreeBSD: too many descriptors. Last segment: 16!

jeremiahL, 2017-05-10:
I'm sometimes getting this message on the console.

too many descriptors. Last segment: 16!

It looks like the dma tag in the driver is being setup wrong. Here is the check that is failing:
```
        if (nsegs > (adapter->max_tx_sgl_size - 2)) {
                device_printf(adapter->pdev,
                    "too many descriptors. Last segment: %d!\n", nsegs);
                for (i = 0; i <= nsegs; i++) { 
                        ena_trace(ENA_WARNING, "frag[%d]: addr:0x%llx, len 0x%x",                           
                            i, (unsigned long long)tx_info->bufs[i].paddr,
                            tx_info->bufs[i].len);
                }

                counter_u64_add(tx_ring->tx_stats.dma_mapping_err, 1);

                rc = ENA_COM_INVAL;
                goto dma_error;
        }
```

You're doing a check for (adapter->max_tx_sgl_size - 2) segments, however when you setup the dma tag, you do not subtract two when passing the max number of segments.

```
        /* Create DMA tag for Tx buffers */
        err = bus_dma_tag_create(bus_get_dma_tag(adapter->pdev),
            1, 0,                       /* alignment, bounds    */
            BUS_SPACE_MAXADDR,          /* lowaddr              */
            BUS_SPACE_MAXADDR,          /* highaddr             */
            NULL, NULL,                 /* filter, filterarg    */
            ENA_TSO_MAXSIZE,            /* maxsize              */
            adapter->max_tx_sgl_size,   /* nsegments            */
            ENA_TSO_MAXSIZE,            /* maxsegsize           */
            0,                          /* flags                */
            NULL,                       /* lockfunc             */
            NULL,                       /* lockfuncarg          */
            &tx_ring->buf_tag);
```

The mismatch is what will cause this error.

gtzalik, 2017-05-11:
jeremiahL, thanks for your report
We will fix it and will upload the fix in the next version

gtzalik, 2017-06-07:
the issue fixed on commit 2bde54f07586d0d5ef0a73cdccd2dfbc33f46b74

closed 2017-06-07 by gtzalik