amzn/amzn-drivers #18: ena driver 1.1.3 failure after a few days of uptime

mfenniak, 2017-05-23:
Hello!  I'm running an i3.4xlarge instance with ENA, on Amazon Linux, and I'm encountering intermittent network outages and ena driver error output.

Originally I was running on Amazon Linux (ECS Optimized) 2016.09.g, which is kernel 4.4.51-40.58.amzn1.x86_64.  After a few times when my i3 instances were terminated by my ASG for failed instance status checks, I disabled termination, and when the issue reoccurred I was able to pull these logs from the machine: https://gist.github.com/mfenniak/56f050b9a4877d1b07388e4d01de829c.  These logs represent a period when the server was unable to use the network for about 20 minutes, but eventually recovered.

I upgraded the same i3 instance to the latest available kernel package (4.9.27-14.31.amzn1.x86_64) and ena driver (1.1.3).  After about 4 days, the same symptoms occurred; a network outage, similar log output: https://gist.github.com/mfenniak/d1bfa2c1c94ee9980113e70ba19ca563.  This time the instance rebooted (cause unknown) and recovered.

It seems suspicious both occurrences that I've logged occurred roughly the same time after boot (335605 - 93 hrs after boot, 365557 - 101 hrs after boot), and in both instances the server would've been under similar load.  I suppose I can expect to see the same issue again in about another 4 days.

Please let me know if there is any additional information I can capture, or any other way I can help diagnose or resolve this issue.

zorikm, 2017-05-23:
Thank you Mathieu,
We are looking into it.

zorikm, 2017-08-28:
Hi Mathieu,
Apologies for delay. I wanted to update you.
We believe that this issue was not triggered by ENA driver, but rather by other component in our infrastructure. It was fixed.
I'm closing this issue, but please feel free to reopen if the problem returns.

mfenniak, 2017-08-29:
Thanks @zorikm.  Will let you know if I see this issue again.

closed 2017-08-29 by mfenniak