amzn/amzn-drivers #19: 4.11 Compile Errors

lconnell, 2017-05-24:
I cannot compile this driver using 4.11 kernel. 4.11.2-1.el7.elrepo.x86_64

```
+ make -C /usr/src/kernels/4.11.2-1.el7.elrepo.x86_64 M=/root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena 'NOSTDINC_FLAGS=-I /root/rpmbuild/BUILD/ena-1.1.3/obj/default/include'
make: Entering directory `/usr/src/kernels/4.11.2-1.el7.elrepo.x86_64'
  LD      /root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena/built-in.o
  CC [M]  /root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena/ena_netdev.o
/root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena/ena_netdev.c: In function 'ena_init_napi':
/root/rpmbuild/BUILD/ena-1.1.3/obj/default/kernel/linux/ena/ena_netdev.c:1493:3: error: implicit declaration of function 'napi_hash_add' [-Werror=implicit-function-declaration]
   napi_hash_add(&adapter->ena_napi[i].napi);
   ^
compilation terminated due to -Wfatal-errors.
```

netanelbelgazal, 2017-05-24:
Thank you Lee.
We are going to upload a new version that fixes this issue within couple of days.
In the meantime you can use our driver from the Linux tree.
Please let me know if this is acceptable as a temporary solution.

lconnell, 2017-05-25:
Thanks for the quick response. Yes, I actually switched to the 4.11 kernel which ships with the 1.1.2 version of the ENA driver.

kwilczynski, 2017-05-25:
Related to https://github.com/amzn/amzn-drivers/issues/14.

netanelbelgazal, 2017-07-18:
Issue fixed by new ENA linux driver.

closed 2017-07-18 by netanelbelgazal