amzn/amzn-drivers #21: Compile errors on kernel 3.2 (Debian 7 Wheezy 64 bits / 3.2.0-4-amd64)

ccastillo-contactpoint, 2017-06-23:
```
Trying to compile the ENA driver on a Debian Wheezy 7.8.aws.1 (ami-e0efab88 / us-east-1) but it fails with this error. Was checking the kcompat.h file and it seems to me that 3.2 kernels are supported on other distros ? Not an expert here so I'm asking if support for this can be added ?

Thanks

###
# Debian 7 Wheezy
# uname -a
# Linux wheezy64-base 3.2.0-4-amd64 #1 SMP Debian 3.2.81-1 x86_64 GNU/Linux
###

root@wheezy64-base:/usr/src/amzn-drivers-1.1.3# dkms add -m amzn-drivers -v 1.1.3

Creating symlink /var/lib/dkms/amzn-drivers/1.1.3/source ->
                 /usr/src/amzn-drivers-1.1.3

DKMS: add completed.
root@wheezy64-base:/usr/src/amzn-drivers-1.1.3# dkms build -m amzn-drivers -v 1.1.3

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=3.2.0-4-amd64 -C kernel/linux/ena/ BUILD_KERNEL=3.2.0-4-amd64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.2.0-4-amd64 (x86_64)
Consult /var/lib/dkms/amzn-drivers/1.1.3/build/make.log for more information.
root@wheezy64-base:/usr/src/amzn-drivers-1.1.3# 
root@wheezy64-base:/usr/src/amzn-drivers-1.1.3# cat /var/lib/dkms/amzn-drivers/1.1.3/build/make.log
DKMS make.log for amzn-drivers-1.1.3 for kernel 3.2.0-4-amd64 (x86_64)
Fri Jun 23 15:29:02 GMT 2017
make: Entering directory `/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena'
make -C /lib/modules/3.2.0-4-amd64/build M=/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/linux-headers-3.2.0-4-amd64'
  CC [M]  /var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena/ena_netdev.h:44:0,
                 from /var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena/kcompat.h:242:8: error: redefinition of ‘struct dev_ext_attribute’
compilation terminated due to -Wfatal-errors.
make[4]: *** [/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena/ena_netdev.o] Error 1
make[3]: *** [_module_/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena] Error 2
make[2]: *** [sub-make] Error 2
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.2.0-4-amd64'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/1.1.3/build/kernel/linux/ena'
root@wheezy64-base:/usr/src/amzn-drivers-1.1.3#
```

netanelbelgazal, 2017-06-28:
Hi Carlos,

Thank you for letting us know.
I'll fix this compilation issue and upload a fix.

Regards,
Netanel

netanelbelgazal, 2017-07-18:
This compilation error fixed by driver version 1.2.0

closed 2017-07-18 by netanelbelgazal