amzn/amzn-drivers #22: FreeBSD: ASSERT panic in RX path when doing scp of a large file

kyenduri, 2017-06-26:
I am hitting a panic with the latest driver bits (commit a5c575083855b01fcaa31d7b3186254bd8d42531).

scp of a large file from my on-prem laptop panicks to the AWS instance triggers the panic. iperf3 from another AWS VM works fine though. So, this looks like a timing related race.

Panic details:

```
ena_rx_mbuf() [TID:100097]: Assert failed on src/freebsd/sys/dev/ena/ena.c:ena_rx_mbuf:1441:Invalid alloc frag buffer

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x1c
fault code              = supervisor write data, page not present
instruction pointer     = 0x820:0xffff860000347017
stack pointer           = 0x828:0xffff8600399998c0
frame pointer           = 0x828:0xffff8600399999c0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = interrupt enabled, resume, IOPL = 0
current process         = 11 (irq258: ena0)
1c: L4 (0) 0
trap number             = 12
panic: page fault
cpuid = 0
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xffff8600399994b0
vpanic() at vpanic+0x114/frame 0xffff8600399994f0
panic() at panic+0x43/frame 0xffff860039999550
trap_fatal() at trap_fatal+0x36b/frame 0xffff8600399995b0
trap_pfault() at trap_pfault+0x171/frame 0xffff8600399995f0
trap() at trap+0x39f/frame 0xffff860039999800
calltrap() at calltrap+0x8/frame 0xffff860039999800
--- trap 0xc, rip = 0xffff860000347017, rsp = 0xffff8600399998d0, rbp = 0xffff8600399999c0 ---
ena_handle_msix() at ena_handle_msix+0x237/frame 0xffff8600399999c0
intr_event_execute_handlers() at intr_event_execute_handlers+0x9e/frame 0xffff860039999a10
ithread_loop() at ithread_loop+0xb6/frame 0xffff860039999a70
fork_exit() at fork_exit+0x84/frame 0xffff860039999ab0
fork_trampoline() at fork_trampoline+0xe/frame 0xffff860039999ab0
```

Looking at the code in ena.c -

```
1680         for (i = 0; i < CLEAN_BUDGET; ++i) {
1681                 rxc = ena_rx_cleanup(rx_ring);
1682 
1683                 /* Protection from calling ena_tx_cleanup from ena_start_xmit */
1684                 ENA_RING_MTX_LOCK(tx_ring);
1685                 txc = ena_tx_cleanup(tx_ring);
1686                 ENA_RING_MTX_UNLOCK(tx_ring);
```

it looks on line 1681, we need some kind of protection. I tried adding the
ENA_RING_MTX_LOCK(rx_ring). It panicked in a different place with that change -

```
panic: Memory modified after free 0xffff86004790c000(16384) val=85006e02 @ 0xffff86004790c000

cpuid = 1
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xffff86003998f690
vpanic() at vpanic+0x114/frame 0xffff86003998f6d0
panic() at panic+0x43/frame 0xffff86003998f730
trash_ctor() at trash_ctor+0x61/frame 0xffff86003998f750
mb_ctor_clust() at mb_ctor_clust+0x18/frame 0xffff86003998f780
uma_zalloc_arg() at uma_zalloc_arg+0x588/frame 0xffff86003998f7f0
m_getjcl() at m_getjcl+0xe3/frame 0xffff86003998f840
ena_refill_rx_bufs() at ena_refill_rx_bufs+0xb3/frame 0xffff86003998f8d0
ena_handle_msix() at ena_handle_msix+0x5b6/frame 0xffff86003998f9c0
intr_event_execute_handlers() at intr_event_execute_handlers+0x9e/frame 0xffff86003998fa10
ithread_loop() at ithread_loop+0xb6/frame 0xffff86003998fa70
fork_exit() at fork_exit+0x84/frame 0xffff86003998fab0
fork_trampoline() at fork_trampoline+0xe/frame 0xffff86003998fab0
```

gtzalik, 2017-06-27:
thanks @kyenduri 
we are working on a fix.

gtzalik, 2017-06-27:
@kyenduri - can you please answer the following to assist us to reproduce it
1. what instance type did you use?
2. what MTU you used (on both sides)?
3. anything else you changed in the network configuration?
4. did you run the scp with special configuration?

Thanks

semihalf-krawczyk-michal, 2017-06-27:
@kyenduri Could you, please, also tell us what version of the FreeBSD are you using? Thanks

kyenduri, 2017-06-27:
@gtzalik 
1.  r4.large
2. AWS: 9001,  remote (on_prem): 1500
3. I have this setting in /boot/policy.conf
net.isr.maxthreads=-1
4. no special scp configuration. The file was around 50MB.

 BTW, I hit this panic this morning, even when the client is another AWS VM. This is with iperf.

@semihalf-krawczyk-michal 
FreeBSD 11.0. My AMI is closest to  public FreeBSD AMI ami-0307a163.

semihalf-krawczyk-michal, 2017-06-29:
We tried to reproduce this issue, however without any success. Could you provide us with few more information?

1. How many iperf3 processes and streams were you using?
2. With what flags iperf3 was executed?
3. Was machine idle during tests, or doing something in the background?
4. Could you provide us with full log? It may be possible that driver was printing additional information few lines earlier (i.e. about resource allocation fail)

Please use for testing unmodified version of driver from this repository. The locking shouldn't be an issue with newest patch (at least on RX), especially if you are not restarting interface during transmission.

Thanks

kyenduri, 2017-06-29:
@semihalf-krawczyk-michal 
I hit the panic only once using iperf3 so far. But, it happens all the time using 'scp' from an on-prem FreeBSD VM. So, please test with the scp case to recreate the panic.

In any case, as for your questions above -
1. One process and one stream
2. server : iperf3 -s 8888 and client iperf3 -c <server-ip> 8888
BTW, the command for scp looks -
 scp <large-file> <aws-vm-public-ip>:/var/tmp
3. The machine is idle.
4. I did not see any driver messages before the panic in /var/log/messages.

semihalf-krawczyk-michal, 2017-06-30:
@kyenduri 
We couldn't reproduce it with scp neither. We were sending GB of data and it never crashed.

Could you assist us with gathering more data to find potential source of this issue?
To enable additional information from the driver, you should uncomment few lines in Makefile of the standalone driver:

```
CFLAGS  += -DENA_DEBUG                                                                                                                

CFLAGS  += -DENA_TRACE                                                                                                                

 
CFLAGS  += -DENA_DEBUG_LEVEL=0x0AF 
#or
CFLAGS  += -DENA_DEBUG_LEVEL=0x0A7  
```                                                                                                 
The value 0x0AF is enabling additional information on RX, however it will slow down the driver a lot and can make this issue unreproducible.
The second value is not showing as detailed info, but we should get some warnings/additional informations. However, the performance will be only slightly affected.

Could you run try to scp the file with one of those setups and send us logs?

Thanks

kyenduri, 2017-07-03:
OK. The problem is at our end. I did not realize ena driver uses software LRO.
The panic turned out to be due to a bad interaction between software LRO and our changes to the RX data path in the networking stack. I no longer see the panic with LRO turned off in the driver.
This would also explain why we did not see it with iperf between 2 AWS VMs. LRO is less likely to kick in for that case.

Thanks for all the help. I am closing the issue.

closed 2017-07-03 by kyenduri