amzn/amzn-drivers #24: RSS hash config support

anakli, 2017-07-03:
I am trying to read/configure the RSS hash key as part of a DPDK application running on a i3.4xlarge AWS instance with enhanced networking support running Ubuntu 16.04. 

The function rte_eth_dev_rss_hash_conf_get() returns an error because dev_ops-> rss_hash_conf_get is null (not supported). Is this expected?

I noticed the [linux kernel ENA driver README](https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena) states that the user can provide the RSS hash function and RSS key via ethtool. However, the device does not seem to support this:

$ sudo ethtool --show-rxfh ens3
Cannot get RX flow hash indir size and/or key size: Operation not permitted

Am I missing something? Which AWS instances (if any) support reading/updating the RSS config?

netanelbelgazal, 2017-07-04:
Amkli,

As of today, ENA instances do allow to configure RSS indirection table.
However, getting indirection table and/or getting/changing RSS key and function are not yet enabled.
(we do plan to enable get RSS configuration on all ENA instances later this year).
Our Linux driver is RSS ready (as mentioned in README file).
As for DPDK, the appropriate support will be added once we have get RSS configuration enabled on instance level.

anakli, 2017-07-04:
Thanks for the information.

closed 2017-07-12 by netanelbelgazal

nirmalanandj, 2018-06-29:
Can someone confirm is this is still the case with the latest ENA driver on Linux on FreeBSD VMs on AWS?

yastreb78, 2018-06-29:
@nirmalanandj , yes this still the case with both Linux and FreeBSD.

nirmalanandj, 2018-06-29:
Do we support the following:
1. Reading the hash key or indirection table entries? Is that in roadmap?
2. Reading if its toeplitz or crc hash?

When can we expect support for configuring custom hash key and indirection table?

yastreb78, 2018-06-29:
All the above is in the roadmap, but unfortunately we can not disclose any timelines.

nirmalanandj, 2018-06-30:
Thanks for the response.Few followup up questions:
Lets say I have an instance with 8 vCPUs. 
1) Can I still configure the ENA device to use only one Rx/Tx queue? 

2) Is asymmetric queues possible. Lets say 'X ' Tx queues and 'Y' Rx queues?  As per  code it looks symmetric, but wanted to know if ENA hardware has such capabilities and can be exposed until our primary requirement of customizing the RSS hash key and indirection table is available. Let us know.

Thanks,
Nirmal

netanelbelgazal, 2018-06-30:
1. Today the driver doesn't support it.
With a slight driver modification you can achieve that.
If you want I can provide a patch that modifies the number of queues.

2. The ENA doesn't allow the user to have asymmetric number of queues.
You have to open them in pairs.

nirmalanandj, 2018-06-30:
Yes a patch could help. Thanks much. Appreciate your quick response. We use FreeBSD. A linux patch also should be good.

talabudi, 2018-08-14:
Hi
I would also want to this feature.
I have 8 cpus and I want to have only 4 active rx/tx queues.
Is there a WA ?
Can I do it on run-time or just on driver load ?

Thanks !

yastreb78, 2018-08-14:
@talabudi if you want to have only 4 queues , I believe that you can change ENA_MAX_NUM_IO_QUEUES to 4 in  https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/common/ena_com/ena_com.h#L54

talabudi, 2018-08-14:
@yastreb78  Thanks !
I meant to control the number of queues at run time.

Another less preferred option is to add a module global parameter and control through modprobe.