amzn/amzn-drivers #26: High retransmits with single connection on newer kernels on 20gbps instances

PaulFurtado, 2017-07-12:
Background:
When we first started using ENA, we were on kernel 3.18 with the first publicly published ENA driver (before a release was tagged). At that time, iperf3 tests showed very few retransmits when using a single connection. Now we have hosts running kernel 4.4 mainline and kernel 4.9 mainline with ena driver version 1.1.3 and we see absurdly high retransmits when using a single connection, but zero retransmits when using multiple parallel connections. I verified that this is fully reproducible on the amazon linux 2017 AMI ami-a4c7edb2 with default settings.

We are running m4.16xlarge instances in placement groups. I suspect that the issue is that a single connection on an m4.16xlarge is capped at 10gbps while the NIC is capped at 20gbps so when using a single connection, linux is flooding the NIC with double the packets it can handle.

### iperf3 output with single connection (kernel 4.9.35, ena 1.1.3):
```
[  4] local 172.18.8.230 port 17684 connected to 172.18.0.223 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  2.33 GBytes  20.0 Gbits/sec  222    227 KBytes       
[  4]   1.00-2.00   sec  1.18 GBytes  10.1 Gbits/sec  6239    210 KBytes       
[  4]   2.00-3.00   sec  1.18 GBytes  10.1 Gbits/sec  6933    122 KBytes       
[  4]   3.00-4.00   sec  1.18 GBytes  10.1 Gbits/sec  6970    297 KBytes       
[  4]   4.00-5.00   sec  1.18 GBytes  10.1 Gbits/sec  6114    157 KBytes       
[  4]   5.00-6.00   sec  1.17 GBytes  10.1 Gbits/sec  6660    218 KBytes       
[  4]   6.00-7.00   sec  1.18 GBytes  10.1 Gbits/sec  6381    122 KBytes       
[  4]   7.00-8.00   sec  1.17 GBytes  10.1 Gbits/sec  6341    122 KBytes       
[  4]   8.00-9.00   sec  1.18 GBytes  10.1 Gbits/sec  6468    175 KBytes       
[  4]   9.00-10.00  sec  1.17 GBytes  10.1 Gbits/sec  6565    218 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  12.9 GBytes  11.1 Gbits/sec  58893             sender
[  4]   0.00-10.00  sec  12.9 GBytes  11.1 Gbits/sec                  receiver

iperf Done.
```

### iperf3 output with `-P 2` (kernel 4.9.35, ena 1.1.3):
```
[  4] local 172.18.8.230 port 17714 connected to 172.18.0.223 port 5201
[  6] local 172.18.8.230 port 17716 connected to 172.18.0.223 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   0.00-1.00   sec  1.20 GBytes  10.3 Gbits/sec    0    856 KBytes       
[SUM]   0.00-1.00   sec  2.41 GBytes  20.7 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   1.00-2.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   1.00-2.00   sec  1.20 GBytes  10.3 Gbits/sec    0    856 KBytes       
[SUM]   1.00-2.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   2.00-3.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   2.00-3.00   sec  1.20 GBytes  10.3 Gbits/sec    0    856 KBytes       
[SUM]   2.00-3.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   3.00-4.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   3.00-4.00   sec  1.20 GBytes  10.3 Gbits/sec    0    856 KBytes       
[SUM]   3.00-4.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   4.00-5.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   4.00-5.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[SUM]   4.00-5.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   5.00-6.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   5.00-6.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[SUM]   5.00-6.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   6.00-7.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   6.00-7.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[SUM]   6.00-7.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   7.00-8.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   7.00-8.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[SUM]   7.00-8.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   8.00-9.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   8.00-9.00   sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[SUM]   8.00-9.00   sec  2.40 GBytes  20.6 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   9.00-10.00  sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[  6]   9.00-10.00  sec  1.20 GBytes  10.3 Gbits/sec    0    900 KBytes       
[... 56 lines cut ...]
```

### iperf3 output when client and server are both on 3.18.36 is roughly the same, minus the initial retransmit spike:
```
[  4] local 172.18.8.230 port 31533 connected to 172.18.0.223 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  2.03 GBytes  17.5 Gbits/sec    0   1.01 MBytes       
[  4]   1.00-2.00   sec  1.46 GBytes  12.5 Gbits/sec  353    245 KBytes       
[  4]   2.00-3.00   sec  1.18 GBytes  10.1 Gbits/sec  314    253 KBytes       
[  4]   3.00-4.00   sec  1.19 GBytes  10.2 Gbits/sec  263    201 KBytes       
[  4]   4.00-5.00   sec  1.17 GBytes  10.1 Gbits/sec  238    288 KBytes       
[  4]   5.00-6.00   sec  1.17 GBytes  10.1 Gbits/sec  307    288 KBytes       
[  4]   6.00-7.00   sec  1.18 GBytes  10.1 Gbits/sec  300    210 KBytes       
[  4]   7.00-8.00   sec  1.17 GBytes  10.1 Gbits/sec  290    114 KBytes       
[  4]   8.00-9.00   sec  1.17 GBytes  10.1 Gbits/sec  226    341 KBytes       
[  4]   9.00-10.00  sec  1.18 GBytes  10.1 Gbits/sec  282    210 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  12.9 GBytes  11.1 Gbits/sec  2573             sender
[  4]   0.00-10.00  sec  12.9 GBytes  11.1 Gbits/sec                  receiver

iperf Done.
```

If I cap iperf3's throughput at 10gbps or use tc to cap the system's throughput at 10gbps, there are zero retransmits on both kernels. I'm wondering what changed that caused the retransmits to get so out of hand. Does something need to be tuned to make the system pace itself at a max of 10gbps per-connection?

netanelbelgazal, 2017-07-12:
Hi Paul,

I will check that and will back to you with an answer.

Regards,
Netanel

PaulFurtado, 2017-08-11:
Hey @netanelbelgazal any update here? 

Thanks!
Paul

netanelbelgazal, 2017-08-13:
Hi Paul,

I reproduce the issue you mention here and I am working on identifying a fix.

Regards,
Netanel

netanelbelgazal, 2017-12-13:
Hi @PaulFurtado 

We fixed the issue you encounter.
The fix is transparent and doesn't require any change from your side.
You should see 0 retransmissions when running iperf.

Please let us know if you still encounter problems.

closed 2017-12-13 by netanelbelgazal