amzn/amzn-drivers #27: Network interface reset due to lost Tx packets

hernandanielg, 2017-07-12:
We faced a network disconnection from our instance that lasted for a few seconds less than 1 minute

### Driver info
```
Jul 12 07:08:09 docker-linux kernel: [20045236.927756] ena: Elastic Network Adapter (ENA) v1.1.2
Jul 12 07:08:09 docker-linux kernel: [20045236.930919] ena 0000:00:03.0: Elastic Network Adapter (ENA) v1.1.2
Jul 12 07:08:09 docker-linux kernel: [20045236.965275] AES CTR mode by8 optimization enabled
Jul 12 07:08:09 docker-linux kernel: [20045237.036805] ena: ena device version: 0.10
Jul 12 07:08:09 docker-linux kernel: [20045237.039665] ena: ena controller version: 0.0.1 implementation version 1
Jul 12 07:08:09 docker-linux kernel: [20045238.496839] ena 0000:00:03.0: creating 8 io queues. queue size: 1024
Jul 12 07:08:09 docker-linux kernel: [20045238.505780] ena 0000:00:03.0: Elastic Network Adapter (ENA) found at mem 83000000, mac addr th:is:is:not:my:mac Queues 8
```

### Issue
Instance had these `soft lockup` issues due to some processes were killed by OOM-killer
```
Jul 12 15:12:04 docker-linux-build-1-dh kernel: [20074223.938094] NMI watchdog: BUG: soft lockup - CPU#48 stuck for 22s! [java:40646]
Jul 12 15:12:04 docker-linux-build-1-dh kernel: [20074251.939111] NMI watchdog: BUG: soft lockup - CPU#48 stuck for 22s! [java:40646]
Jul 12 15:12:04 docker-linux-build-1-dh kernel: [20074291.940437] NMI watchdog: BUG: soft lockup - CPU#48 stuck for 23s! [java:40646]
```

Checking `/var/log/syslog` we found these kind of messages complaining about `Tx that wasn't completed on time` (times are in UTC)
```
Jul 12 15:12:04 docker-linux kernel: [20074195.605296] ena 0000:00:03.0 eth0: Found a Tx that wasn't completed on time, qid 5, index 0.
Jul 12 15:12:04 docker-linux kernel: [20074195.605298] ena 0000:00:03.0 eth0: Found a Tx that wasn't completed on time, qid 5, index 1.
...
Jul 12 15:12:04 docker-linux kernel: [20074195.605867] ena 0000:00:03.0 eth0: Found a Tx that wasn't completed on time, qid 5, index 1022.
Jul 12 15:12:04 docker-linux kernel: [20074195.605871] ena 0000:00:03.0 eth0: Found a Tx that wasn't completed on time, qid 5, index 1023.

```

also it was complaining about `The number of lost tx completion is above the threshold` and this was the action that triggered the reset
```
Jul 12 15:12:04 docker-linux kernel: [20074195.605531] ena 0000:00:03.0 eth0: The number of lost tx completion is above the threshold (129 > 128). Reset the device                                                                                                  
Jul 12 15:12:04 docker-linux kernel: [20074195.605535] ena 0000:00:03.0 eth0: The number of lost tx completion is above the threshold (130 > 128). Reset the device
...
Jul 12 15:12:04 docker-linux kernel: [20074195.605869] ena 0000:00:03.0 eth0: The number of lost tx completion is above the threshold (230 > 128). Reset the device                                                                                                  
Jul 12 15:12:04 docker-linux kernel: [20074195.605872] ena 0000:00:03.0 eth0: The number of lost tx completion is above the threshold (231 > 128). Reset the device
```

The bit of reset was set to the interface according to this line
https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ena_netdev.c#L2849

Finally interface was reset 
`Jul 12 15:12:04 docker-linux-build-1-dh kernel: [20074195.606002] ena 0000:00:03.0 eth0: Trigger reset is on
`
   
We want to know why this happens and how to avoid this to happen again

Let me know if I am missing any information here

zorikm, 2017-07-12:
Hi,
There are soft lockups in your system, and most probably, as a result ENA interrupt processing for Queue #5 is starved (CPU is unavailable due to soft lockup). The ENA driver has a built-in self-recovery mechanism that resets the interface in order to heal the situation.

closed 2017-07-19 by netanelbelgazal

UdayShankar517, 2020-06-29:
Could you explain how to avoid this issue ?

akiyano, 2020-06-29:
@UdayShankar517 Please see if the  detailed instructions I posted in https://github.com/amzn/amzn-drivers/issues/84 help you

thenralmani, 2021-01-21:
We are having the same issue with latest ENA driver 2.4.0g. any conclusion on this reset error, how to fix/avoid - thanks

ena  ens5: Found a Tx that wasn't completed on time,
ens5: The number of lost tx completions is above the threshold (372 > 128). Reset the device
Driver info: Elastic Network Adapter (ENA) v2.4.0g

akiyano, 2021-01-22:
@hernandanielg Please see my previous response, which redirects you to read my explanation in https://github.com/amzn/amzn-drivers/issues/84

pparth, 2021-03-29:
What can we do so that the OOM killer does not cause this? The support team told us to increase the memory for the container that got killed so that OOM should essentially never happen, which is not a valid answer!

AWSNB, 2021-03-29:
@pparth could you share instance-id so we could look at logs from AWS firmware side?

pparth, 2021-03-29:
You can find all the details in case id 8119390641. It would be useful that support people do not provide such lame answers please. We run more than 50 containers in a host and, obviously to us, we need to protect the host and the other containers from a container that goes rogue and consume all the memory without setting a hard limit.

pparth, 2021-03-29:
Are there any characteristics of the process that gets killed by OOM, that we need to avoid so that the network on the host does not go off? For example, we see that the problem is exaggerated when the process has hundreds of children sleep processes and this may cause tons of info to be dumped when the parent is killed.

AWSNB, 2021-03-29:
Typically this happens when cpu core is so busy that it can’t give cycles to the driver Tx napi to process transmit completion queue

One specific case which make this happen more often is in case of excessive output to serial port that end up starving core0

Recommendation mitigation:

  1.  Reduce output to console
And/or

  1.  Move network interrupt process affinity of core0

We’ll review the logs Monday morning PDT and update u

arthurk85, 2021-03-29:
Hi @pparth,

Let's continue this discussion in case id 8119390641

Thanks,
Arthur

akiyano, 2021-03-29:
Hi @pparth,

Let's continue this discussion in case id 8119390641

Thanks,
Arthur