amzn/amzn-drivers #29: FreeBSD: missing TSO support

kyenduri, 2017-07-13:
I am running FreeBSD 11.1-RC2 #0 on an AWS r4.large instance and see the following in the 'options' field -

ena0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 1500
	options=422<TXCSUM,JUMBO_MTU,LRO>

I don't see TSO in the 'options' field. I see in the README file -
The ENA driver supports:
- TSO over IPv4/IPv6

Is this a device limitation or the FreeBSD ena driver limitation?

semihalf-krawczyk-michal, 2017-07-14:
The FreeBSD driver have TSO support implemented, however the device on AWS isn't supporting it, yet.

kyenduri, 2017-07-14:
Thanks! Do you know when we will get TSO support for the device in AWS?

kyenduri, 2017-07-15:
Also, I see that there is no RXCSUM support. This is causing software LRO to be skipped. Specifically, tcp_lro_rx() is not called in the driver because of the lack of RXCSUM support. This could have a performance impact.

Is the lack of RXCSUM a device limitation?

netanelbelgazal, 2017-07-19:
Rx checksum is currently unsupported by the device.
We plan to support it in future releases.

gtzalik, 2017-07-24:
@kyenduri - As Michal and Netanel mentioned above, the ENA device is currently doesn't support Rx checksum and TSO. We do plan to add these features in the future but I don't have estimate time for releasing them now.
closing this issue. let me know if you need anything else.

closed 2017-07-24 by gtzalik

kyenduri, 2017-07-24:
Thanks!  I wanted to confirm this is not a FreeBSD driver specific issue and got the confirmation. Linux driver also has the same behavior.

erenon, 2024-10-17:
Does the current AWS NIC support TSO now?

akiyano, 2024-10-17:
Hi @erenon,
The ENA device currently does not support TSO.