amzn/amzn-drivers #37: error: nested redefinition of 'enum pkt_hash_types'

cgianelloni-applause, 2017-09-13:
This is on CentOS 7.3.1611 w/ kernel `3.10.0-693.2.2.el7.x86_64` whereas kernel `3.10.0-514.26.2.el7.x86_64` worked.

```make -C /lib/modules/3.10.0-693.2.2.el7.x86_64/build M=/opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena modules
              make[1]: Entering directory `/usr/src/kernels/3.10.0-693.2.2.el7.x86_64'
                CC [M]  /opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena/ena_netdev.o
              In file included from /opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena/ena_netdev.h:44:0,
                               from /opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena/ena_netdev.c:53:
              /opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena/kcompat.h:382:6: error: nested redefinition of 'enum pkt_hash_types'
               enum pkt_hash_types {
                    ^
              compilation terminated due to -Wfatal-errors.
              make[2]: *** [/opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena/ena_netdev.o] Error 1
              make[1]: *** [_module_/opt/applause/src/amzn-drivers-ena_linux_1.1.3/kernel/linux/ena] Error 2
              make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.2.2.el7.x86_64'
              make: *** [all] Error 2```

jhaohai, 2017-09-14:
same thing here for centos 7.4

yastreb78, 2017-09-14:
Thank you for the report.
This is going to be handled as part of new version 1.3 which will be released soon.

netanelbelgazal, 2017-09-19:
ENA version 1.3.0 is out.
It contains the fix for your issue.

closed 2017-09-19 by netanelbelgazal

nethershaw, 2017-09-20:
@netanelbelgazal With respect, it does not.

I see that a fix has been applied for the RPM makefile, but the fix for kernel/linux/ena/kcompat.h has _not_ been applied for RHEL_RELEASE_VERSION(7,4).

netanelbelgazal, 2017-09-24:
@nethershaw 

You are right.
I didn't push the commit to master, I only pushed the tag.
Please check now.

nethershaw, 2017-09-27:
@netanelbelgazal 

Thank you. Builds as expected.