amzn/amzn-drivers #39: error: wrong stats when using ena_get_stats (kernel < 2.6.36)

azrle, 2017-09-14:
Hi,

I observed that stats get wrong when using ena-driver 1.2.0 in kernel 2.6.32 (CentOS 6).

```
$ ethtool -i eth0
driver: ena
version: 1.2.0g
firmware-version:
bus-info: 0000:00:03.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

$ cat /proc/net/dev; sudo ethtool -S eth0 | grep tx_bytes
Inter-|   Receive                                                |  Transmit
 face |bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls carrier compressed
    lo: 6919841279 29352140    0    0    0     0          0         0 6919841279 29352140    0    0    0     0       0          0
  eth0: 6225205581 376664102    0    0    0     0          0         0 20661073610 284794167    0    0    0     0       0          0
     queue_0_tx_bytes: 4087952612
     queue_1_tx_bytes: 6640832540
     queue_2_tx_bytes: 4582622571
     queue_3_tx_bytes: 5800756290
     queue_4_tx_bytes: 4048643657
     queue_5_tx_bytes: 4257765549
     queue_6_tx_bytes: 4023754776
     queue_7_tx_bytes: 4398625609
```
eth0 TX bytes should be about 37.84G according to `ethtool`. However, `/proc/net/dev` shows that the TX bytes are 20.66G.

It seems that `bytes`, `packets` and `rx_drops` in `ena_get_stats`  have wrong types.  IMHO, They should be `unsigned long` instead of `unsigned int`.
http://elixir.free-electrons.com/linux/v2.6.32.71/source/include/linux/netdevice.h#L128

kernel >= 2.6.36 is not affected since `ena_get_stats64` is used.

I have confirmed that the problem is fixed after following patching.
```diff
--- a/kernel/linux/ena/ena_netdev.c
--
+++ b/kernel/linux/ena/ena_netdev.c
@@ -2545,20 +2545,20 @@ static struct net_device_stats *ena_get_stats(struct net_device *netdev)
{
struct ena_adapter *adapter = netdev_priv(netdev);
struct ena_ring *rx_ring, *tx_ring;
-       unsigned int rx_drops;
+       unsigned long rx_drops;
struct net_device_stats *stats = &netdev->stats;
unsigned int start;
int i;
 
memset(stats, 0, sizeof(*stats));
for (i = 0; i < adapter->num_queues; i++) {
-               unsigned int  bytes, packets;
+               unsigned long  bytes, packets;
 
tx_ring = &adapter->tx_ring[i];
do {
start = u64_stats_fetch_begin_irq(&tx_ring->syncp);
-                       packets = (unsigned int)tx_ring->tx_stats.cnt;
-                       bytes = (unsigned int)tx_ring->tx_stats.bytes;
+                       packets = (unsigned long)tx_ring->tx_stats.cnt;
+                       bytes = (unsigned long)tx_ring->tx_stats.bytes;
} while (u64_stats_fetch_retry_irq(&tx_ring->syncp, start));
 
stats->tx_packets += packets;
@@ -2568,8 +2568,8 @@ static struct net_device_stats *ena_get_stats(struct net_device *netdev)
 
do {
start = u64_stats_fetch_begin_irq(&tx_ring->syncp);
-                       packets = (unsigned int)rx_ring->rx_stats.cnt;
-                       bytes = (unsigned int)rx_ring->rx_stats.bytes;
+                       packets = (unsigned long)rx_ring->rx_stats.cnt;
+                       bytes = (unsigned long)rx_ring->rx_stats.bytes;
} while (u64_stats_fetch_retry_irq(&tx_ring->syncp, start));
 
stats->rx_packets += packets;
@@ -2578,7 +2578,7 @@ static struct net_device_stats *ena_get_stats(struct net_device *netdev)
 
do {
start = u64_stats_fetch_begin_irq(&tx_ring->syncp);
-               rx_drops = (unsigned int)adapter->dev_stats.rx_drops;
+               rx_drops = (unsigned long)adapter->dev_stats.rx_drops;
} while (u64_stats_fetch_retry_irq(&tx_ring->syncp, start));
 
stats->rx_dropped = rx_drops;
```

zorikm, 2017-09-14:
Thanks @azrle!
Nice catch!
We'll review and merge it.
Zorik

netanelbelgazal, 2017-09-19:
Thanks for your help.
Your change was merged to driver version 1.3.0

closed 2017-09-19 by netanelbelgazal