amzn/amzn-drivers #40: Error Installing ENA networking kernel driver under Debian Jessie

joeweoj, 2017-09-26:
Hi.

While building a Debian Jessie image using [bootstrap-vz](https://github.com/andsens/bootstrap-vz) i hit an issue installing ENA network drivers.

Bootstrap-vz pulls amzn-drivers master branch so i hit this while using (currently [this rev](https://github.com/amzn/amzn-drivers/commit/b0eff7b0287db28432c473e76ec557a108567f44)).

Here is the output from bootstrap-vz:

> 2017-09-26 09:29:00,185 - bootstrapvz.base.tasklist.run - INFO - Installing ENA networking kernel driver using DKMS
> 2017-09-26 09:29:00,233 - bootstrapvz.common.tools.root.git.handle_stderr - ERROR - Cloning into '/target/5521ed29/root/usr/src/amzn-drivers-1.0.0'...
> 2017-09-26 09:29:07,728 - bootstrapvz.common.tools.root.chroot.handle_stderr - ERROR - Error! Bad return status for module build on kernel: 3.16.0-4-amd64 (x86_64)
> 2017-09-26 09:29:07,728 - bootstrapvz.common.tools.root.chroot.handle_stderr - ERROR - Consult /var/lib/dkms/amzn-drivers/1.0.0/build/make.log for more information.

Contents of `make.log`

> root@box:~# chroot /target/cae4cc9e/root
> root@box:/# cat /var/lib/dkms/amzn-drivers/1.0.0/build/make.log
> DKMS make.log for amzn-drivers-1.0.0 for kernel 3.16.0-4-amd64 (x86_64)
> Tue 26 Sep 10:01:05 UTC 2017
> make: Entering directory '/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'
> make -C /lib/modules/3.16.0-4-amd64/build M=/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena modules
> make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-amd64'
> make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-amd64'
>   CC [M]  /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o
> /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_probe’:
> /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.c:3480:3: error: implicit declaration of function ‘devm_ioremap_wc’ [-Werror=implicit-function-declaration]
>    ena_dev->mem_bar = devm_ioremap_wc(&pdev->dev,
>    ^
> compilation terminated due to -Wfatal-errors.
> cc1: some warnings being treated as errors
> /usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:262: recipe for target '/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o' failed
> make[4]: *** [/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o] Error 1
> /usr/src/linux-headers-3.16.0-4-common/Makefile:1354: recipe for target '_module_/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena' failed
> make[3]: *** [_module_/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena] Error 2
> Makefile:181: recipe for target 'sub-make' failed
> make[2]: *** [sub-make] Error 2
> Makefile:8: recipe for target 'all' failed
> make[1]: *** [all] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-4-amd64'
> Makefile:19: recipe for target 'all' failed
> make: *** [all] Error 2
> make: Leaving directory '/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'
> root@box:/#

I have resolved this issue (purely to unblock me) by hardcoding a different version of the amzn-drivers. By default it seems to pull master, but by pinning it to [ena_linux_1.2.0](https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_1.2.0).

This issue has been raised under bootstrap-vz [here](https://github.com/andsens/bootstrap-vz/issues/399).

Thanks

netanelbelgazal, 2017-09-27:
Thank you for informing us.
I'll check the issue and report back soon.

netanelbelgazal, 2017-10-02:
@joeweoj,

I have a fix for the problem you mention.
I'll push the patch soon.
Meanwhile if you would like, you can apply it onto your repository and report if you see any additional issues.

```diff --git a/kernel/linux/ena/kcompat.h b/kernel/linux/ena/kcompat.h
index 8e29faa364bd..443ab66d2f43 100644
--- a/kernel/linux/ena/kcompat.h
+++ b/kernel/linux/ena/kcompat.h
@@ -546,7 +546,8 @@ static inline void napi_complete_done(struct napi_struct *n, int work_done)
 #endif
 
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0) \
-       || (UBUNTU_VERSION_CODE && UBUNTU_VERSION_CODE >= UBUNTU_VERSION(3,13,0,126))
+       || ((UBUNTU_VERSION_CODE && UBUNTU_VERSION_CODE >= UBUNTU_VERSION(3,13,0,126)) && \
+          (UBUNTU_VERSION_CODE && UBUNTU_VERSION_CODE < UBUNTU_VERSION(3,14,0,0)))
 #else
 
 static inline void ioremap_release(struct device *dev, void *res)

valdemon, 2017-10-09:
@netanelbelgazal : confirm that the patch does the job.

netanelbelgazal, 2017-10-10:
@valdemon 
Thank you for your confirmation.

bcorijn, 2017-11-08:
@netanelbelgazal any estimation when this fix could get pushed/merged?

netanelbelgazal, 2017-11-12:
@bcorijn 
The fix will be pushed in the next day or two.

netanelbelgazal, 2017-11-14:
@bcorijn 
I just push a new release that should fix this issue.
If you still run into trouble, don't hesitate to reopen this issue.

closed 2017-11-14 by netanelbelgazal