amzn/amzn-drivers #41: ena interface keep restarting on freebsd

yemga, 2017-10-03:
I have two freebsd instances, both are same OS version and kernel versions. but unfortunately one instance ena adapter is keep restarting every one minute, help is much appreciated.

Oct  2 19:59:25 nfs2 kernel: ena0: ENA admin queue is not in running state!
Oct  2 19:59:25 nfs2 kernel: ena0: Trigger reset is on
Oct  2 19:59:25 nfs2 kernel: ena0: device is going DOWN
Oct  2 19:59:26 nfs2 kernel: ena0: Allocated msix_entries, vectors (cnt: 9)
Oct  2 19:59:26 nfs2 kernel: ena0: ena0: device is going UP
Oct  2 19:59:26 nfs2 kernel: ena0: link is UP
Oct  2 19:59:26 nfs2 kernel: queue 0 - cpu 8
Oct  2 19:59:26 nfs2 kernel: ena0: queue 1 - cpu 9
Oct  2 19:59:26 nfs2 kernel: ena0: queue 2 - cpu 10
Oct  2 19:59:26 nfs2 kernel: ena0: queue 3 - cpu 11
Oct  2 19:59:26 nfs2 kernel: ena0: queue 4 - cpu 12
Oct  2 19:59:26 nfs2 kernel: ena0: queue 5 - cpu 13
Oct  2 19:59:26 nfs2 kernel: ena0: queue 6 - cpu 14
Oct  2 19:59:26 nfs2 kernel: ena0: queue 7 - cpu 15
Oct  2 19:59:33 nfs2 kernel: ena0: ENA admin queue is not in running state!
Oct  2 19:59:33 nfs2 kernel: ena0: Trigger reset is on
Oct  2 19:59:33 nfs2 kernel: ena0: device is going DOWN
Oct  2 19:59:34 nfs2 kernel: ena0: Allocated msix_entries, vectors (cnt: 9)
Oct  2 19:59:34 nfs2 kernel: ena0: ena0: device is going UP
Oct  2 19:59:34 nfs2 kernel: ena0: link is UP
Oct  2 19:59:34 nfs2 kernel: queue 0 - cpu 0
Oct  2 19:59:34 nfs2 kernel: ena0: queue 1 - cpu 1
Oct  2 19:59:34 nfs2 kernel: ena0: queue 2 - cpu 2
Oct  2 19:59:34 nfs2 kernel: ena0: queue 3 - cpu 3
Oct  2 19:59:34 nfs2 kernel: ena0: queue 4 - cpu 4
Oct  2 19:59:34 nfs2 kernel: ena0: queue 5 - cpu 5
Oct  2 19:59:34 nfs2 kernel: ena0: queue 6 - cpu 6
Oct  2 19:59:34 nfs2 kernel: ena0: queue 7 - cpu 7

FreeBSD nfs2 11.1-RELEASE-p1 FreeBSD 11.1-RELEASE-p1 #0: Wed Aug  9 11:55:48 UTC 2017     root@amd64-builder.daemonology.net:/usr/obj/usr/src/sys/GENERIC  amd64

gtzalik, 2017-11-06:
Thanks @yemga 
Can you please retest with the latest driver (version 0.8.0)?
Thanks

gtzalik, 2017-12-14:
@yemga - did you try the newer version?

closed 2017-12-18 by gtzalik