amzn/amzn-drivers #43: Compiling ENA driver on RHEL 6.5 with 3.16.49 kernel fails

ahmedelz, 2017-10-24:
Hi,

I'm trying to compile kernel/linux/ena driver on RHEL 6.5 with 3.16.49 kernel on an EC2 instance but it fails with:

```
  CC [M]  /usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.o
/usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.c: In function ‘ena_probe’:
/usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.c:3480: error: implicit declaration of function ‘devm_ioremap_wc’
```

I see the function devm_ioremap_wc is defined at the bottom of kcompat.h if KERNEL_VERSION is less than 4.1.0 but that doesn't seem to be working for my case. linux/version.h has:

`#define LINUX_VERSION_CODE 200753`

ahmedelz, 2017-10-24:
`uname -r` returns `3.16.49` and `rpm -qa kernel-devel` returns `kernel-devel-3.16.49-1.x86_64`

netanelbelgazal, 2017-10-24:
@ahmedelz
Can you please provide us the ami-id of your instance?

ahmedelz, 2017-10-24:
The AMI ID is: ami-591dc623

netanelbelgazal, 2017-10-25:
This AMI is a private AMI.
Can you please share it with:
952466867617

ahmedelz, 2017-10-25:
Ok. I shared it with the account number provided

netanelbelgazal, 2017-10-26:
I still have issues connecting to your instance.
It require password.
Let's continue this discussion over email.
my email - netanel@amazon.com

netanelbelgazal, 2017-11-14:
@ahmedelz 
The new driver I just pushed fixes your issue.

closed 2017-11-14 by netanelbelgazal