amzn/amzn-drivers #44: Download version 1.3.0 but got 1.0.2 after compiled

a9tm, 2017-11-02:
Hi,
I downloaded release 1.3.0 file from this github, after following instruction, I only got 1.0.2 instead of 1.3.0

Download link: https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_1.3.0

OS installed: CentOS 7 (1708)
Steps:

    # yum install gcc kernel-devel-$(uname -r)
    # cd amzn-drivers-ena_linux_1.3.0/kernel/linux/ena
    # make
    # insmod ena.ko
    # modinfo ena

```
filename:       /lib/modules/3.10.0-693.5.2.el7.x86_64/kernel/drivers/net/ethernet/amazon/ena/ena.ko.xz
version:        1.0.2
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
rhelversion:    7.4
srcversion:     0ADEBA934369F8D450E5CE4
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:
intree:         Y
vermagic:       3.10.0-693.5.2.el7.x86_64 SMP mod_unload modversions
signer:         CentOS Linux kernel signing key
sig_key:        C7:57:A9:FB:BD:0D:82:C9:E5:40:52:02:9A:09:08:D1:7C:F1:AD:C7
sig_hashalgo:   sha256
parm:           debug:Debug level (0=none,...,16=all) (int)
```

Where did I get wrong?

yastreb78, 2017-11-02:
You still keep using your original driver
You should be copying  ena.ko to /lib/modules/$(uname -r)/  and running "sudo depmod" , followed by restart.
After restart you should see new driver running

netanelbelgazal, 2017-11-14:
@a9tm,
Since you got an answer I am closing this issue.
Let us know if you need any further assistance.

closed 2017-11-14 by netanelbelgazal