amzn/amzn-drivers #45: DPDK ena driver performance issue

anakli, 2017-11-09:
I measured 10 Gb/s network bandwidth between two i3.large instances using iperf (i.e., using the Linux kernel ena driver for the NIC). However, when I run a DPDK application such as [dpdk-iperf](https://github.com/ansyun/dpdk-iperf) (using the DPDK ena driver for the NIC), I only see up to 2.5 Gb/s bandwidth. 

What is the expected per-core throughput achievable with the DPDK ena driver? Is there a particular benchmark you recommend running to verify proper DPDK setup on EC2? 

I am using DPDK v17.05 and I have applied the patches provided in this repo. I have ensured that my EC2 instances are in the same placement group. However, I am only getting 2.5 Gb/s instead of 10 Gb/s. Any suggestions?

yastreb78, 2017-11-09:
@anakli 

DPDK driver should be achieving same BW as Linux driver (assuming same conditions). We will double check the specific application behavior, but one thing that comes up right away - MTU.

In linux (when running in AWS) you will have default MTU to be set to JUMBO (9001 bytes). However, in dpdk-ans (that dpdk-iperf seems to be using) JUMBO mode is nott set  (https://github.com/ansyun/dpdk-ans/blob/master/ans/ans_main.c#L178) and default 1.5K packets seems to be used.

anakli, 2017-11-14:
Thanks, I tried running the dpdk-iperf test with dpdk-ans option `--enable-jumbo --max-pkt-len=9000` but I am still seeing the same throughput of ~2.5 Gb/s. I am running a dpdk-iperf3 server and a Linux iperf3 client.

The application I ultimately want to run on EC2 is a remote storage server called [ReFlex] (https://github.com/stanford-mast/reflex/tree/userspace). When I run ReFlex on a bare-metal cluster, I can easily saturate 10 Gb/s. However, running the same code gives me only up to 2.5 Gb/s on EC2 -- the same network bottleneck I observed with dpdk-iperf. 

In terms of DPDK NIC configuration in rte_eth_conf.rxmode, I currently have `jumbo_frame=1`, `hw_ip_checksum=1` and `hw_strip_crc=1`. Are there any other offloads that need to be configured to achieve 10 Gb/s with ENA?

Which benchmark do you typically use to test DPDK performance? I would be curious to run a benchmark you use to achieve 10 Gb/s, since I would like to rule out instance configuration issues. I have configured each instance with a secondary Ethernet interface (which I use for DPDK) and I assigned a Elastic IP addr for the primary interface (which I use to ssh into the machine). Is this the recommended approach for DPDK setup on EC2? I have confirmed that when I use the *kernel* ena driver for the interfaces, iperf shows 10 Gb/s between primary and secondary interfaces. The issue is only when I use the DPDK driver.

Thanks for your help.

netanelbelgazal, 2017-11-14:
The configuration you used is the recommended one.
Can you give me more information about the instance type and the image you are using?

I also advise we'll continue this discussion over email.
My email is netanel@amazon.com

anakli, 2017-11-14:
I am using i3.large with Amazon Linux. Thanks, I will follow up over email.

Luo-Liang, 2018-05-31:
Any follow up on this?

I just checked the latest version of ANS and on C5 instances all I get is 3Gbps/s while the default Linux ena driver gives 9.6Gbps

gtzalik, 2018-06-06:
@Luo-Liang - We are still working on the testing tool that we will be able to share with you.
It should be ready soon and I will update you as soon as it will be released to get your feedback.

aloknnikhil, 2019-12-09:
Bumping the thread up. I have multiple DPDK apps that have the same issue. I just cannot exceed 4 Gbps. Is there an active issue here? I start losing packets if I push more than 4 Gbps

zorikm, 2019-12-10:
@aloknnikhil,
Thanks for reaching out. We'll need more information. What instance type are you using? What kind of traffic is this? If it is a TCP or UDP traffic how many streams are you using?   Is this a traffic within VPC? If so, do you use placement groups?

aloknnikhil, 2019-12-15:
> @aloknnikhil,
> Thanks for reaching out. We'll need more information. What instance type are you using? What kind of traffic is this? If it is a TCP or UDP traffic how many streams are you using? Is this a traffic within VPC? If so, do you use placement groups?

I'm using a c5n.9xlarge. I have 2 ENIs that are being dedicated for TX and RX. This is UDP traffic generated by Cisco's TRex (https://trex-tgn.cisco.com/) using a single flow/stream and thus a single queue. This traffic is all within the same VPC. I did have the instance in a cluster placement group but didn't make a difference and I didn't expect it to either (since the source and the sink for the traffic is the same instance). Additionally, I was able to drive ~9 Gbps using the Linux kernel stack using iPerf between the 2 ENIs.

I-gor-C, 2019-12-16:
> I'm using a c5n.9xlarge. I have 2 ENIs that are being dedicated for TX and RX. This is UDP traffic of packet size 1400 generated by Cisco's TRex (https://trex-tgn.cisco.com/) using a single flow/stream and thus a single queue. This traffic is all within the same VPC. I did have the instance in a cluster placement group but didn't make a difference and I didn't expect it to either (since the source and the sink for the traffic is the same instance). Additionally, I was able to drive ~9 Gbps using the Linux kernel stack using iPerf between the 2 ENIs.

Hi @aloknnikhil 
A few more questions for clarity:
1. The 4 Gbps throughput, is it measured on the Tx side?
2. What DPDK version are you using?
3. The 9 Gbps throughput with Linux kernel stack - what parameters are you setting? Is it also single UDP flow with 1400-byte-sized packets?

aloknnikhil, 2019-12-16:
> Hi @aloknnikhil
> A few more questions for clarity:
>
> 1. The 4 Gbps throughput, is it measured on the Tx side?
> 2. What DPDK version are you using?
> 3. The 9 Gbps throughput with Linux kernel stack - [...]

1 - It was measured on the RX. The TX was close to 8 Gbps. The rest of the packets were dropped.
2 - 19.05
3 - Yes. Same packet size. Specifically, this is the command I used 
``` iperf3 -c <IP> -u -b 0 -l 1400 -P 1 ```

I-gor-C, 2019-12-17:
> 1 - It was measured on the RX. The TX was close to 8 Gbps. The rest of the packets were dropped.
> 2 - 19.05
> 3 - Yes. Same packet size. Specifically, this is the command I used
> `iperf3 -c <IP> -u -b 0 -l 1400 -P 1`

Thanks, @aloknnikhil 
We might need some non-public info to continue the investigation from AWS side - let's continue via emails. My email is igorch@amazon.com, I'll send you a message in a minute.

I-gor-C, 2020-04-22:
Closing this issue, please reach out to us again if happens again.

closed 2020-04-22 by I-gor-C