amzn/amzn-drivers #46: Unable to load ena module in freebsd

mtmrm, 2017-11-10:
`FreeBSD c12-nc1c-OPNsense-111.0-RELEASE-p12 FreeBSD 11.0-RELEASE-p12 #0 02581be96(stable/17.7): Sat Aug 26 11:00:39 CEST 2017     root@sensey64:/usr/obj/usr/src/sys/SMP  amd64
`
This is the uname -a output. This doesn't provide a kernel revision number, so I went ahead and compiled using stable kernel sources. (This is a freebsd installation boostrapped to have opnsense, a fork of pfsense on top of it.) The compiled module fails to load saying `KLD if_ena.ko: depends on kernel - not available or version mismatch
linker_load_file: Unsupported file type
`
Googling around, I found that it might be due to the kern.osreldate and the BSD_VERSION in param.h. I modified the param.h file located at /usr/src/sys/sys/param.h and at /usr/include/sys/param.h to reflect the sysctl value and recompiled. It still fails, throwing the same error. Any help/ guidance is appreciated. I have not been able to find a way to force load that module.

Thanks.

gtzalik, 2017-11-10:
Thanks @mtmrm 
We will check this issue and will get back to you soon.

semihalf-krawczyk-michal, 2017-11-14:
Hi @mtmrm
Do you have access to the sources from which the kernel was built?
You can use the revision number only if the kernel was built directly from svn FreeBSD kernel tree. If the kernel was built from other sources (there it seems to be built from git repository and commit 02581be96), you will have to built the module using them.

The process will be similar, except you will have to checkout to the given commit
```
git clone sources_link
cd source_dir
git checkout 02581be96

# and then you can:
export SYSDIR=`pwd`/sys
# or
mkdir /usr/src/
cp * /usr/src/
```

gtzalik, 2017-12-04:
Hi @mtmrm
Did the above instructions help to solve your issue?

gtzalik, 2017-12-14:
@mtmrm - I assume the issue was resolved. please reopen if you still need help

closed 2017-12-14 by gtzalik