amzn/amzn-drivers #49: release tag 1.5 points to wrong commit

akemner, 2017-12-14:
release tag https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_1.5.0 points to https://github.com/amzn/amzn-drivers/commit/85e34d6b734c7defb93bced8998a831290f61a43

but this doesnt contain some changes that are in the commit in master like the changes to kernel/linux/rpm

i believe the correct release tag for 1.5.0 should be
https://github.com/amzn/amzn-drivers/commit/8770d557df1681e5a7923ba052ab246d65acca88

netanelbelgazal, 2017-12-20:
@akemner, 
You are right.
I fixed the broken tag.

closed 2017-12-20 by netanelbelgazal

akemner, 2017-12-20:
@netanelbelgazal the tag still looks broken to me

netanelbelgazal, 2017-12-24:
@akemner Did you remove the old tag from your repository before you tried to fetch?
When I clone a newly fresh repo I can see ena_linux_1.5.0 points to the right place.