amzn/amzn-drivers #50: Release notes and stable version

piyush-kansal, 2017-12-28:
Hi,

Could you please start adding release notes to the ENA driver releases so that one can make an informed decision about whether to upgrade to the latest releases or not?

Besides, adding a stable tag or something would also be of great help.

Thanks.

netanelbelgazal, 2017-12-29:
Hi @piyush-kansal,

We already have release notes in the ENA driver.
Direct link: https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/RELEASENOTES.md

piyush-kansal, 2017-12-29:
Oops. My bad. Thanks for your quick reply though.

Is it safe to assume that the most recent version of the drivee is also the most stable one?

netanelbelgazal, 2017-12-29:
Yes,
All the releases in the Github are stable.

closed 2017-12-29 by piyush-kansal