amzn/amzn-drivers #51: DPDK ENA PMD - TX L4 offload flags set in RX path

mgsmith1000, 2018-01-05:
When using the ENA PMD with an application that does IPsec tunneling, I have encountered a problem. Some packets (UDP or TCP) that are sent over an IPsec tunnel fail the integrity check at the receiving side. Other IP protocols (ICMP) are not affected, they pass the integrity check and are successfully received. I debugged and found that the following is occurring:

- When a UDP or TCP packet arrives on an ENA interface, ena_rx_mbuf_prepare() sets PKT_TX_TCP_CKSUM or PKT_TX_UDP_CKSUM on ol_flags for the packet mbuf.
- The DPDK AESNI MB cryptodev encrypts the original packet buffer in place. An ESP header and IP header are prepended in the packet headroom.
- The modified packet is sent out an ENA interface. Since PKT_TX_TCP_CKSUM or PKT_TX_UDP_CKSUM is set on ol_flags for the packet mbuf, ena_tx_mbuf_prepare() enables hardware L4 checksum generation for the L4 protocol of the original unencrypted packet.
- The ENA hardware generates a TCP or UDP checksum and writes it at an appropriate offset for a TCP or UDP packet. Since the next header after the IP header is now an ESP header (instead of original TCP or UDP), this ends up writing the an L4 checksum into a location that is not valid for the modified packet. If the original packet was a UDP packet, the lower 16 bits of the ESP sequence number are overwritten. If the original packet was a TCP packet, the TCP checksum is written into the payload containing the padded/encrypted version of the packet.
- The packet is forwarded to the IPsec tunnel peer, which calculates an ICV using the ESP header & encrypted packet. The hash doesn’t match the received ICV, which was generated based on the packet contents prior to their modification by the hardware, so the packet is dropped.

This seems to be unique to the ENA PMD. The same behavior does not occur when tunneling packets over IPsec using the same application running against some other NIC types (ixgbe, i40e, e1000). I have not tested other types of tunnels (IP-in-IP, GRE, etc), but it seems like a very good possibility that packets that are sent over those types of tunnels could also be affected.

A patch similar to the following resolves the issue for the application running in my tests:
```
diff --git a/drivers/net/ena/ena_ethdev.c b/drivers/net/ena/ena_ethdev.c
index 22db895..6f982f6 100644
--- a/drivers/net/ena/ena_ethdev.c
+++ b/drivers/net/ena/ena_ethdev.c
@@ -261,16 +261,6 @@ static inline void ena_rx_mbuf_prepare(struct rte_mbuf *mbuf,
 {
 	uint64_t ol_flags = 0;
 
-	if (ena_rx_ctx->l4_proto == ENA_ETH_IO_L4_PROTO_TCP)
-		ol_flags |= PKT_TX_TCP_CKSUM;
-	else if (ena_rx_ctx->l4_proto == ENA_ETH_IO_L4_PROTO_UDP)
-		ol_flags |= PKT_TX_UDP_CKSUM;
-
-	if (ena_rx_ctx->l3_proto == ENA_ETH_IO_L3_PROTO_IPV4)
-		ol_flags |= PKT_TX_IPV4;
-	else if (ena_rx_ctx->l3_proto == ENA_ETH_IO_L3_PROTO_IPV6)
-		ol_flags |= PKT_TX_IPV6;
-
 	if (unlikely(ena_rx_ctx->l4_csum_err))
 		ol_flags |= PKT_RX_L4_CKSUM_BAD;
 	if (unlikely(ena_rx_ctx->l3_csum_err))
```

Would it be possible to make a change to the PMD code similar to the above that leaves the TX offload flags unset when the packet is received? This would allow the application to decide whether to set the TX checksum offload flags. If it is necessary to store information about the L3/L4 protocols of a received packet so the application can use that information while processing the packet, the mbuf packet_type could be used for that purpose.

gtzalik, 2018-01-09:
@mgsmith1000 We will investigate it and will let you know soon.
Thanks

gtzalik, 2018-01-21:
@mgsmith1000 - I'm closing this issue as it was resolved.
Please feel free to contact us if you have further issues.

closed 2018-01-21 by gtzalik