amzn/amzn-drivers #54: fail to compile on centos 6.6

aflat, 2018-01-18:
```
# uname -r
2.6.32-504.30.3.el6.x86_64

# yum list kernel-devel
Loaded plugins: fastestmirror, presto
Loading mirror speeds from cached hostfile
 * base: centos.aol.com
 * epel: s3-mirror-us-east-1.fedoraproject.org
 * extras: centos.mirror.constant.com
 * updates: bay.uchicago.edu
Installed Packages
kernel-devel.x86_64                                                                  2.6.32-504.30.3.el6                                                                   installed
Available Packages
kernel-devel.x86_64                                                                  2.6.32-696.18.7.el6                                                                   updates

# cat /etc/redhat-release
CentOS release 6.6 (Final)
```

I've tried using make, and rpmbuilding the src rpm, same issue.
```
make -C /usr/src/kernels/2.6.32-504.30.3.el6.x86_64 M=/root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena 'NOSTDINC_FLAGS=-I /root/rpmbuild/BUILD/ena-1.5.0/obj/default/include'
make: Entering directory `/usr/src/kernels/2.6.32-504.30.3.el6.x86_64'
  LD      /root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/built-in.o
  CC [M]  /root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/ena_netdev.o
/root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/ena_netdev.c: In function 'ena_calc_io_queue_num':
/root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/ena_netdev.c:3286: error: implicit declaration of function 'pci_msix_vec_count'
compilation terminated due to -Wfatal-errors.
make[1]: *** [/root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/ena_netdev.o] Error 1
make: *** [_module_/root/rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena] Error 2
make: Leaving directory `/usr/src/kernels/2.6.32-504.30.3.el6.x86_64'
error: Bad exit status from /var/tmp/rpm-tmp.yJ2QIl (%build)
```

aflat, 2018-01-19:
I upgraded the kernel to 2.6.32-642.11.1.el6.x86_64 and compile went fine. Not sure what the lowest level kernel that is supported, but you might want to throw in some flags to fail out when it's too old.

netanelbelgazal, 2018-01-21:
Hi,
2.6.32-504 should also compile, I'll take care of it.

akiyano, 2018-03-01:
Hi @aflat, 

Please fetch the latest version of the driver, it should now compile on CentOs 6.6.

Regards,
Arthur

closed 2018-03-06 by akiyano