amzn/amzn-drivers #55: CentOS 7.4 complaining non-stop in /var/log/messages

genio, 2018-02-09:
Nothing special has been done to install ena support other than updating to the latest release of CentOS 7.4.

```bash
[root@machine ~]# modinfo ena
filename:       /lib/modules/3.10.0-693.17.1.el7.x86_64/kernel/drivers/net/ethernet/amazon/ena/ena.ko.xz
version:        1.0.2
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
rhelversion:    7.4
srcversion:     3A6B9F1766C9A0B5CBC7D01
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:        
intree:         Y
vermagic:       3.10.0-693.17.1.el7.x86_64 SMP mod_unload modversions 
signer:         CentOS Linux kernel signing key
sig_key:        50:6C:68:68:80:9D:2C:BF:54:0B:F0:D9:83:D5:C6:70:9D:BC:4F:22
sig_hashalgo:   sha256
parm:           debug:Debug level (0=none,...,16=all) (int)
[root@webwest2 ~]# uname -a
Linux machine.domain 3.10.0-693.17.1.el7.x86_64 #1 SMP Thu Jan 25 20:13:58 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
[root@machine ~]# cat /etc/redhat-release 
CentOS Linux release 7.4.1708 (Core) 
[root@machine ~]# ethtool -i eth0
driver: ena
version: 1.0.2
firmware-version: 
expansion-rom-version: 
bus-info: 0000:00:05.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
[root@machine ~]#  tail -f /var/log/messages
Feb  8 21:50:32 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:35 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:38 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:41 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:44 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:47 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:50 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:53 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:56 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:50:59 machine kernel: ena: Feature 27 isn't supported
Feb  8 21:51:02 machine kernel: ena: Feature 27 isn't supported

```

zorikm, 2018-02-09:
Hi @genio,
This is a very old driver. I would recommend you to install the latest 1.5.0.
The message itself is harmless: looks like periodical invocation of ethtool get_settings() method that wasn't supported in 1.0.2 ena driver.

peterborghard, 2018-02-15:
Does anyone know the best practices for upgrading the driver, I'm running into the same issue.  I've been troubleshooting some network performance issues with cloud-init/s3, can't tell yet if this is related.  I've upgraded network drivers before by vendor, never for aws "ena".

davdunc, 2018-03-01:
@peterborghard The typical method is to leverage the [DKMS](https://github.com/dell/dkms) to create an RPM supporting your kernel. See [Enabling Enhanced Networking with ENA on Ubuntu](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html#enhanced-networking-ena-linux) where the DKMS build process is outlined.

solarsnake, 2018-03-07:
This issue still appears even when upgrading to the latest driver set from amazon. Has anyone found a way to suppress the message? Even with logrotate zipping these daily it's a bit aggressive.

akiyano, 2018-03-08:
Hi @solarsnake,

Your driver upgrade was probably unsuccessful as the bug is fixed in all versions in this git repository.
I suggest we continue this discussion offline. 
Please contact me via email - akiyano@amazon.com 

Best regards,
Arthur

closed 2018-03-26 by akiyano

ElijahLynn, 2018-04-10:
@solarsnake Have you confirmed the version you are on with `modinfo ena`? We are getting these Feature 27 messages with 1.0.2 which is not in this repo and we need to upgrade.

cirocosta, 2018-07-12:
Hi @zorikm , thanks for the insights!

One more question - would you deem relevant an update from 1.0.2 to 1.5.3 for more reasons other than getting rid of such warning? Are there performance improvements and some other interesting features that only a recent driver would leverage?

Thanks in advance,
Ciro

zorikm, 2018-07-12:
Hi @cirocosta,
Absolutely. 1.0.2 is very outdated. Multiple issues were found and fixed since then. I certainly recommend you to update your instances with the latest 1.5.3.

cirocosta, 2018-07-12:
@zorikm, Thanks very much!

ElijahLynn, 2018-07-12:
To confirm, the latest update for our ORG, which was ena 1.2.0, did remove these errors from our logs.