amzn/amzn-drivers #56: Compilation error with 4.15 kernel

scotte, 2018-02-26:
Compiling this driver on 4.15 kernel results in the following error:

```
  CC [M]  /var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.o
/var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_refill_rx_bufs’:
/var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.c:544:12: error: ‘__GFP_COLD’ undeclared (first use in this function); did you mean ‘__GFP_COMP’?
            __GFP_COLD | GFP_ATOMIC | __GFP_COMP);
            ^~~~~~~~~~
            __GFP_COMP
compilation terminated due to -Wfatal-errors.
```

__GCP_COLD has been removed from the kernel: https://github.com/torvalds/linux/commit/453f85d43fa9ee243f0fc3ac4e1be45615301e3f

scotte, 2018-02-26:
Next error:

```
  CC [M]  /var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.o
/var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_probe’:
/var/lib/dkms/ena/1.5.0/build/kernel/linux/ena/ena_netdev.c:3669:2: error: implicit declaration of function ‘setup_timer’; did you mean ‘sk_stop_timer’? [-Werror=implicit-function-declaration]
  setup_timer(&adapter->timer_service, ena_timer_service,
  ^~~~~~~~~~~
  sk_stop_timer
compilation terminated due to -Wfatal-errors.
```

scotte, 2018-02-26:
I'm going to leave replacing setup_timer() with timer_setup() to someone with more kernel knowledge than I...

SerialVelocity, 2018-02-27:
@scotte This is probably what you are looking for:
https://github.com/torvalds/linux/commit/e99e88a9d2b067465adaa9c111ada99a041bef9a#diff-cbb0d5dcb5420a41759e0295d2af2ae9

and the _GFP_COLD was fixed directly in the kernel by:
https://github.com/torvalds/linux/commit/453f85d43fa9ee243f0fc3ac4e1be45615301e3f#diff-cbb0d5dcb5420a41759e0295d2af2ae9

akiyano, 2018-02-27:
Hi @scotte,

Thanks for the proposed fixes.
We will be releasing a patch in the next few days that will fix all of the above problems.

Best regards,
Arthur

scotte, 2018-02-27:
Thank you @akiyano and @SerialVelocity

scotte, 2018-03-01:
With a06c07d643e91e3a7f982ff1d4eefaa01dccc22f (version 1.5.1) the driver compiles on 4.15. Thank you!

closed 2018-03-01 by scotte