amzn/amzn-drivers #60: Cannot turn on adaptive-rx

sargun, 2018-03-13:
I get an error turning on adaptive rx:
```
# ethtool -C eth4 adaptive-rx on
Cannot set device coalesce parameters: Operation not supported
```

Info about module:
```
# cat /sys/module/ena/version 
1.5.0g
```

I don't see anything in sysfs related to this:
```
# find /sys|grep intr_moderation
# cat /boot/config-$(uname -r)|grep ^CONFIG_SYSFS=
CONFIG_SYSFS=y
```

zorikm, 2018-03-13:
Hello Sargun,
Thank you for your comments.
Adaptive interrupt coalescing is not supported.
We believe that ENA device interrupt generation heuristics coupled with Linux NAPI mechanism provides sufficient control over interrupt rate.
Apologies for not up-to-date Readme. We'll update it to reflect this.

gtzalik, 2019-03-20:
@sargun, I'm closing this issue. please re-open if anything else is needed

closed 2019-03-20 by gtzalik