amzn/amzn-drivers #63: Instances not starting as c5.* with wheezy

choseh, 2018-03-19:
3.2.0-4-amd64 #1 SMP Debian 3.2.96-2 x86_64 GNU/Linux

filename:       /lib/modules/3.2.0-4-amd64/updates/dkms/ena.ko
version:        1.5.1g
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
srcversion:     121BE61723081CBF841045A
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:        
vermagic:       3.2.0-4-amd64 SMP mod_unload modversions 
parm:           debug:Debug level (0=none,...,16=all) (int)

unfortunately I can't provide additional logs, since the instances just plain hang and i have to force stop them and --no-ena-support them to work again

please let me know if there's a way to provide additional debug output

zorikm, 2018-03-19:
Hello @choseh 
May we ask you to provide your region and instance id?

Thanks

choseh, 2018-03-19:
hi @zorikm I tried with i-e01d602e and yesterday with i-c8cf1a09 in eu-central-1

zorikm, 2018-03-19:
Looks like you are using C4 instance type for your current instance and T2 for the previous one.
Both of these instance types DO NOT support ENA.
Please see https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking.html for the list of instance types that support ENA.

Thanks.

choseh, 2018-03-19:
@zorikm I had to change the instance type back again, due to the instances hanging obviously. I'll set one to c5 and enable ena, so you can have a look.

zorikm, 2018-03-19:
Your issue doesn't seem to be related to ENA. 
We suspect that your AMI doesn't suit C5 instance type.
Please contact AWS support.

Thanks

zorikm, 2018-03-19:
@choseh, 
To open EC2 support request,  in your console, on the upper right corner select support center, and then you will see "create case".

closed 2018-05-27 by akiyano