amzn/amzn-drivers #64: Build failed for RedHat RHEL 7.5 beta kernel

svpcom, 2018-04-03:
Build failed for RedHat RHEL 7.5 beta kernel:
``` .sh
$ make
make -C /lib/modules/3.10.0-830.el7.x86_64/build M=/home/fg/amzn-drivers/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-830.el7.x86_64'
  CC [M]  /home/fg/amzn-drivers/kernel/linux/ena/ena_netdev.o
In file included from /home/fg/amzn-drivers/kernel/linux/ena/ena_netdev.h:44:0,
                 from /home/fg/amzn-drivers/kernel/linux/ena/ena_netdev.c:53:
/home/fg/amzn-drivers/kernel/linux/ena/kcompat.h:416:6: error: nested redefinition of 'enum pkt_hash_types'
 enum pkt_hash_types {
      ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [/home/fg/amzn-drivers/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/home/fg/amzn-drivers/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-830.el7.x86_64'
make: *** [all] Error 2
```

```
$ rpm -qip kernel-3.10.0-830.el7.x86_64.rpm 
warning: kernel-3.10.0-830.el7.x86_64.rpm: Header V3 RSA/SHA256 Signature, key ID fd431d51: NOKEY
Name        : kernel
Version     : 3.10.0
Release     : 830.el7
Architecture: x86_64
Install Date: (not installed)
Group       : System Environment/Kernel
Size        : 64339119
License     : GPLv2
Signature   : RSA/SHA256, Mon Jan 15 21:45:08 2018, Key ID 199e2f91fd431d51
Source RPM  : kernel-3.10.0-830.el7.src.rpm
Build Date  : Mon Jan 15 17:47:58 2018
Build Host  : x86-040.build.eng.bos.redhat.com
Relocations : (not relocatable)
Packager    : Red Hat, Inc. <http://bugzilla.redhat.com/bugzilla>
Vendor      : Red Hat, Inc.
URL         : http://www.kernel.org/
Summary     : The Linux kernel
Description :
The kernel package contains the Linux kernel (vmlinuz), the core of any
Linux operating system.  The kernel handles the basic functions
of the operating system: memory allocation, process allocation, device
input and output, etc.
```

akiyano, 2018-04-03:
Hi @svpcom,

Thanks for the report.
We are working on it. Will update you regarding expected delivery date.

Regards,
Arthur

akiyano, 2018-04-10:
Hi @svpcom,

A fix to this problem will be included in the next release.
Until then you can use the following patch:
[0001-Compilation-temporary-fix-for-RHEL7.5-compilation-er.patch.txt](https://github.com/amzn/amzn-drivers/files/1896361/0001-Compilation-temporary-fix-for-RHEL7.5-compilation-er.patch.txt)

Regards,
Arthur

Petenerd, 2018-04-19:
Any idea when the next release will be?

netanelbelgazal, 2018-04-22:
@Petenerd 
We are going to have a release in the upcoming week.

mornap, 2018-04-30:
Any updates / ETA on this issue? We're currently broke on RHEL 7.5 and would rather not code in a fix just to remove it later.

akiyano, 2018-05-03:
Hi @svpcom @Petenerd @mornap,

Version 1.5.3 was just released. 
It contains the fix you need for this issue.

Best regards,
Arthur

closed 2018-05-10 by akiyano