amzn/amzn-drivers #66: cannot build amzn-drivers for Ubuntu 3.13.0-29-generic

ambirag, 2018-04-23:
I was following this steps to enable ENA on Ubuntu AWS instance - https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html

:~# uname -a
Linux ip-x-x-x-x 3.13.0-29-generic #53-Ubuntu SMP Wed Jun 4 21:00:20 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

It failed at **Step 7 - Sub step b**

**Error:**
`root@ip-x-x-x-x:~# sudo dkms build -m amzn-drivers -v 1.0.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=3.13.0-29-generic -C kernel/linux/ena/ BUILD_KERNEL=3.13.0-29-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for amzn-drivers: 1.0.0 not found
Error! Bad return status for module build on kernel: 3.13.0-29-generic (x86_64)
Consult /var/lib/dkms/amzn-drivers/1.0.0/build/make.log for more information.`

**Full log**:  /var/lib/dkms/amzn-drivers/1.0.0/build/make.log
`DKMS make.log for amzn-drivers-1.0.0 for kernel 3.13.0-29-generic (x86_64)
Mon Apr 23 12:28:40 UTC 2018
make: Entering directory `/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'
make -C /lib/modules/3.13.0-29-generic/build M=/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-29-generic'
  CC [M]  /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o
/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.c: In function ‘ena_set_rx_hash’:
/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.c:1052:22: error: storage size of ‘hash_type’ isn’t known
  enum pkt_hash_types hash_type;
                      ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-29-generic'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'
`

akiyano, 2018-04-25:
Hi @ambirag, 

Thanks for the report.
We are working on it. Will update you regarding a fix.

Regards,
Arthur

ambirag, 2018-04-26:
thanks @akiyano

akiyano, 2018-05-10:
Hi @ambirag,

Sorry for the holdup.
Attached is a temporary patch you can use to solve the problem.

[0001-Compilation-Fix-kcompat-fix-compilation-error-in-ubu.patch.txt](https://github.com/amzn/amzn-drivers/files/1990926/0001-Compilation-Fix-kcompat-fix-compilation-error-in-ubu.patch.txt)

To make the driver please run (regular make without the parameter will not work):
make ENA_LEGACY_LINUX=1

A fix will be included in future releases. I will update you here when it happens.

Regards,
Arthur

ambirag, 2018-05-29:
thanks @akiyano , I tried your path 2 weeks ago, but though patch was updated successfully, still one of the aws step for enabling ENA failed, I found it easy to upgrade to latest Ubuntu so I just did that.

sudo do-release-upgrade ( took around 30mins and requires manual input)

Thanks again for your support!