amzn/amzn-drivers #67: cannot build drivers

cristiansavaro, 2018-04-25:
I am trying to build the drivers for oracle 3.10.0-862.el7.x86_64 kernel.
Here is the error i get for 1.5.0 , 1.5.1 and 1.5.2 :  

```
+ echo 'Building for kernel: 3.10.0-862.el7.x86_64 flavors: '\''default'\'''
Building for kernel: 3.10.0-862.el7.x86_64 flavors: 'default'
+ echo 'Build var: kmodtool = /root/amzn-drivers/kernel/linux/rpm/kmodtool'
Build var: kmodtool = /root/amzn-drivers/kernel/linux/rpm/kmodtool
+ echo 'Build var: kverrel = 3.10.0-862.el7.x86_64'
Build var: kverrel = 3.10.0-862.el7.x86_64
+ for flavor in default
+ rm -rf obj/default
+ cp -r source obj/default
+ symvers=source/Module.symvers-x86_64
+ '[' -e source/Module.symvers-x86_64 ']'
++ '[' default = default ']'
+ make -C /usr/src/kernels/3.10.0-862.el7.x86_64 M=/root/amzn-drivers/kernel/lin                                                                                                                                                             ux/rpm/build/ena-1.5.1/obj/default/kernel/linux/ena 'NOSTDINC_FLAGS=-I /root/amz                                                                                                                                                             n-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/default/include'
make[1]: Entering directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
  LD      /root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/default/kernel                                                                                                                                                             /linux/ena/built-in.o
  CC [M]  /root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/default/kernel                                                                                                                                                             /linux/ena/ena_netdev.o
In file included from /root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/de                                                                                                                                                             fault/kernel/linux/ena/ena_netdev.h:44:0,
                 from /root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/de                                                                                                                                                             fault/kernel/linux/ena/ena_netdev.c:53:
/root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/default/kernel/linux/ena                                                                                                                                                             /kcompat.h:414:6: error: nested redefinition of 'enum pkt_hash_types'
 enum pkt_hash_types {
      ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [/root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/default/ke                                                                                                                                                             rnel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/root/amzn-drivers/kernel/linux/rpm/build/ena-1.5.1/obj/de                                                                                                                                                             fault/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
error: Bad exit status from /var/tmp/rpm-tmp.vGNbCF (%build)

```

akiyano, 2018-04-25:
Hi @cristiansavaro,

This is the same as https://github.com/amzn/amzn-drivers/issues/64
Please use the same patch I provided there:  
[0001-Compilation-temporary-fix-for-RHEL7.5-compilation-er.patch.txt](https://github.com/amzn/amzn-drivers/files/1945913/0001-Compilation-temporary-fix-for-RHEL7.5-compilation-er.patch.txt)
A fix will be included in the next release.

Regards,
Arthur

cristiansavaro, 2018-04-25:
@akiyano thanks. i will try this patch today.

cristiansavaro, 2018-04-25:
the patch is successful but the error persists.

akiyano, 2018-04-25:
Hi @cristiansavaro ,

I've tested this patch on the kernel version you have.
I suggest we continue this discussion offline.
Please contact me via email - akiyano@amazon.com

Best regards,
Arthur

akiyano, 2018-05-03:
Patching is no longer necessary, the fix is included in version 1.5.3.

petronazeh, 2018-08-17:
guys i have note 3 n9005 htle i installed nethunter on aryamod rom modified stock touch wiz and i have mt7601 wifi adapter it shows lsusb that the adapter connected but when i set it up shows nothing so i must install drivers and compile my 3.4.111 nethuner kerenal so how i can run that adapter on my note

akiyano, 2018-08-19:
Hi @petronazeh,

Here we deal with issues with the ENA driver.
You are asking a question that is unrelated to us.
We can't help you with that. 

Arthur