amzn/amzn-drivers #71: ena-1.5.3 driver with 2.6.32-696.23.1 kernel crashes in dev_watchdog under iperf load

gakimov, 2018-05-07:
I've build ena-1.5.3 kernel module for our 2.6.32-696.23.1 CentOS 6 kernel and tested it under iperf-2.0.9 between two AWS m5.large instances.
Client finished normally:
./iperf -c 10.40.2.203
------------------------------------------------------------
Client connecting to 10.40.2.203, TCP port 5001
TCP window size:  640 KByte (default)
------------------------------------------------------------
[  3] local 10.40.2.7 port 50764 connected with 10.40.2.203 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-11.8 sec  1.02 GBytes   748 Mbits/sec

Server "crashed" / fell out of network with:

May  7 17:31:03 src@ip-10-40-2-203 -bash: bash[3451]: [root] ./iperf -s
May  7 17:31:25 src@ip-10-40-2-203 kernel: ------------[ cut here ]------------
May  7 17:31:25 src@ip-10-40-2-203 kernel: WARNING: at net/sched/sch_generic.c:265 dev_watchdog+0x1f1/0x200() (Not tainted)
May  7 17:31:25 src@ip-10-40-2-203 kernel: Hardware name: m5.large
May  7 17:31:25 src@ip-10-40-2-203 kernel: NETDEV WATCHDOG: eth0 (ena): transmit queue 1 timed out
May  7 17:31:25 src@ip-10-40-2-203 kernel: Modules linked in: iptable_filter ip_tables video output sbs sbshc ipmi_msghandler i2c_piix4 serio_raw parport_pc i2c_core parport ena(U) dm_crypt drbg ansi_cprng aesni_intel ablk_helper cryptd lrw aes_i586 aes_generic xts gf128mul ctr cbc ecb dm_mod nvme ext3 jbd mbcache [last unloaded: scsi_wait_scan]
May  7 17:31:25 src@ip-10-40-2-203 kernel: Pid: 0, comm: swapper Not tainted 2.6.32-696.23.1.qal2_0.i686 #1
May  7 17:31:25 src@ip-10-40-2-203 kernel: Call Trace:
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0463c49>] ? warn_slowpath_common+0x89/0xe0
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c07ecb71>] ? dev_watchdog+0x1f1/0x200
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0463d33>] ? warn_slowpath_fmt+0x33/0x50
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c07ecb71>] ? dev_watchdog+0x1f1/0x200
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0492de0>] ? ktime_get+0x70/0x150
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0474956>] ? run_timer_softirq+0x136/0x2c0
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c07ec980>] ? dev_watchdog+0x0/0x200
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c046b2db>] ? __do_softirq+0xab/0x200
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c046b46d>] ? do_softirq+0x3d/0x50
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c046b7f5>] ? irq_exit+0x65/0x70
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0434a15>] ? smp_apic_timer_interrupt+0x35/0x50
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0886f7d>] ? apic_timer_interrupt+0x31/0x38
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c043e9a2>] ? native_safe_halt+0x2/0x10
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c041241d>] ? default_idle+0x4d/0xc0
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c0408aa8>] ? cpu_idle+0x98/0xe0
May  7 17:31:25 src@ip-10-40-2-203 kernel: [<c087f5f0>] ? start_secondary+0x2df/0x33c
May  7 17:31:25 src@ip-10-40-2-203 kernel: ---[ end trace 52ae61bbbebcc7a4 ]---
May  7 17:31:25 src@ip-10-40-2-203 kernel: ena 0000:00:05.0: eth0: Transmit time out
May  7 17:36:28 src@ip-10-40-2-203 tagger[2640]: clean_reboot@2640: Executing Shutdown
May  7 17:36:28 src@ip-10-40-2-203 acpid.event: Power button pressed
May  7 17:36:28 src@ip-10-40-2-203 init: Switching to runlevel: 0

zorikm, 2018-05-07:
Hello George,
Thanks for notifying us.
Have you modified the driver?
I'm asking because usually we expect more ENA messages in dmesg following netdev timeout event.
Also, what's your instance-id and what region are you running in?

Thanks!
Zorik

gakimov, 2018-05-07:
No, I have not modified the driver source code, only packaged it in our own style rpm. I can share the srpm with you if needed.
i-0e52d09e0b139086d and i-0dc283b79daeeff18 in us-east-1c. Both currently powered down; if you need ssh access to it, we'll need to work out the procedure somehow. There are DEV builds of Qualys network security appliances.

zorikm, 2018-05-07:
Dear George,
Those are m4.l instances. m4.l instances do not use ENA device. If you'd like to use ENA device on M4 family instances, you should switch to m4.16xl.
Please take a look at https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html for the list of instance types that support ENA.
Please let me know if you have further questions.

Thanks,
Zorik

gakimov, 2018-05-07:
Thanks, I know that. I was testing different kinds of network adapters in the same instance. It was c5.large at the time I opened this Issue and it is c5.large now.

zorikm, 2018-05-07:
Not sure I'm following. The instances you provided above are m4.l.
If you had an issue with c5.large please provide its instance-id and also kernel logs.

Thanks,
Zorik

zorikm, 2018-05-07:
Got it.
George, 
I suggest to continue investigation offline and update the Issue with our findings.
May I ask you to send us (zorik@amazon.com and akiyano@amazon.com) the full kernel log, and also if possible more details about what has been done and how loaded the instance was when you were running iperf.
The driver provides a callback for netdev WD event. This callback dumps the important stats to the kernel log and resets the device in order to recover. From the logs you have provided it looks like this callback wasn't invoked (we also do not see a recovery attempt in our infrastructure logs).
We'd like to understand why ndo_tx_timeout callback hasn't been invoked.
https://elixir.bootlin.com/linux/v2.6.32.69/source/net/sched/sch_generic.c#L258

Thanks,
Zorik

akiyano, 2018-10-25:
Problem was solved with client. closing.

closed 2018-10-25 by akiyano