amzn/amzn-drivers #72: error: nested redefinition of âenum pkt_hash_typesâ  kernel 3.10.0-862.el7.x86_64 (x86_64)

operations-at-alison, 2018-05-10:
CentOs
yum update
# cat /var/lib/dkms/amzn-drivers/1.0.0/build/make.log
DKMS make.log for amzn-drivers-1.0.0 for kernel 3.10.0-862.el7.x86_64 (x86_64)
Thu May 10 08:56:08 UTC 2018
make: Entering directory `/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'
make -C /lib/modules/3.10.0-862.el7.x86_64/build M=/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
  CC [M]  /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.h:44:0,
                 from /var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.c:53:
/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/kcompat.h:424:6: error: nested redefinition of âenum pkt_hash_typesâ
 enum pkt_hash_types {
      ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
make: *** [all] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/1.0.0/build/kernel/linux/ena'

akiyano, 2018-05-10:
Hi @operations-at-alison,

This bug was already fixed with the release of version 1.5.3. (https://github.com/amzn/amzn-drivers/commit/857b4b8d81a1b3ca2a4af29695acda57cabee9cc)
I double checked with your exact kernel (3.10.0-862.el7.x86_64).
Please make sure you use the latest version of the driver.

Best regards,
Arthur

operations-at-alison, 2018-05-11:
Als,o I've updated CentOs to kernel 3.10.0-862.2.3.el7.x86_64 with the same result.

I will try to do a procedure with instructions on https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena page.
Best Regards

Milos

closed 2018-05-27 by akiyano