amzn/amzn-drivers #74: Cannot build ena driver on stock Debian 9 AMI

Senso, 2018-05-29:
After following the README:

```
~/amzn-drivers/kernel/linux/ena $ sudo make
make -C /lib/modules/4.9.0-6-amd64/build M=/home/manuel/amzn-drivers/kernel/linux/ena modules
make[1]: Entering directory '/lib/modules/4.9.0-6-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.0-6-amd64/build'
Makefile:36: recipe for target 'all' failed
make: *** [all] Error 2
```

The `/lib/modules/4.9.0-6-amd64/build` directory did not exit, I created it by hand. 

I'm trying to upgrade the ena module to 1.5.x, since my instance current runs 1.0.2 and syslog is being _heavily_ spammed with lines like this:
`kernel: ena: Feature 27 isn't supported`

akiyano, 2018-05-29:
Hi @Senso,

You are missing the linux-headers package which is needed for compiling the module.
To get this package please run the following command: sudo apt-get install linux-headers-$(uname -r)
After this you will be able to compile the driver.

For full driver installation process please follow steps 1-9 in "Enabling Enhanced Networking with ENA on Ubuntu" here: https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking-ena.html#enhanced-networking-ena-linux

If the command in stage 8 doesn't work, run the following command instead: 
sudo update-initramfs -u -k all

If the command in stage 9 doesn't work, run it with sudo.

Regards,
Arthur

closed 2018-05-29 by akiyano

Senso, 2018-05-30:
Hi, thanks for the reply. I'm afraid that it's not working either, after installing the needed linux-headers packages. 

If I try to compile the ena module by hand, I still get this error:
`make[1]: *** No rule to make target 'modules'.  Stop.`

Same problem when I follow the instructions by using dkms:

```
:~ $ sudo dkms build -m amzn-drivers -v 1.5.3 --kernelsourcedir /usr/src/linux-headers-4.9.0-6-amd64

Preparing kernel 4.9.0-6-amd64 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make mrproper...(bad exit status: 2)
Warning: using /usr/src/linux-headers-4.9.0-6-amd64/.config
(I hope this is the correct config for this kernel)
make oldconfig...(bad exit status: 2)
make prepare-all...(bad exit status: 2)

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=4.9.0-6-amd64 -C kernel/linux/ena/ BUILD_KERNEL=4.9.0-6-amd64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.0-6-amd64 (x86_64)
Consult /var/lib/dkms/amzn-drivers/1.5.3/build/make.log for more information.
```

The content of that log file goes back to `no rules to make target 'module'`:
```
:~ $ cat /var/lib/dkms/amzn-drivers/1.5.3/build/make.log
DKMS make.log for amzn-drivers-1.5.3 for kernel 4.9.0-6-amd64 (x86_64)
Wed May 30 01:24:54 UTC 2018
make: Entering directory '/var/lib/dkms/amzn-drivers/1.5.3/build/kernel/linux/ena'
make -C /lib/modules/4.9.0-6-amd64/build M=/var/lib/dkms/amzn-drivers/1.5.3/build/kernel/linux/ena modules
make[1]: Entering directory '/var/lib/dkms/amzn-drivers/1.5.3/build/kernel/linux/ena'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.0-6-amd64/build'
Makefile:36: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/var/lib/dkms/amzn-drivers/1.5.3/build/kernel/linux/ena'
```

Senso, 2018-05-30:
Ah, I had to replace the symlink `/lib/modules/4.9.0-6-amd64/source` from `/usr/src/linux-headers-4.9.0-6-common` to `/usr/src/linux-headers-4.9.0-6-amd64` and that seems to have fixed it, the module builds with dkms.

rumir98, 2019-03-27:
Could you tell me what command did you use to do it?

zorikm, 2019-03-27:
ln  -s  /usr/src/linux-headers-4.9.0-6-amd64  /lib/modules/4.9.0-6-amd64/source
should do it