amzn/amzn-drivers #75: Porting driver from FreeBSD to DragonFlyBSD

BradHoffman96, 2018-06-04:
Hello!

I don't necessarily have an issue, just needed a way of starting a conversation in order to ask some questions about the FreeBSD implementation of the ENA driver. I am very new to both BSD systems as well, so there might be some ignorant questions or comments. 

Anyway, I am trying to get a better understanding of how the FreeBSD implementation uses the OS's built-in RSS capabilities when running. Is there any other usage of the RSS capabilities aside from setting the pkthdr flowid and the rsstype that I should know about when porting this?

semihalf-krawczyk-michal, 2018-06-05:
Hi Brad,

Setting flowid and RSS type should be enough. 

The FreeBSD is also having special RSS API which was used in older ENA version (inside `#ifdef RSS_SUPORT`). However, it is not fully finished, so it cannot be used with ENA for now.

BradHoffman96, 2018-06-06:
Can you give me a little more detail into the whereabouts of the hash function? I've read through the ena_com.h code and looked at the ena_com_set_hash_function and get_function commands, but I see no indicator of an actual hash function? Is this abstracted into FreeBSD?

gtzalik, 2018-10-25:
@BradHoffman96 This issue was already resolved as far as I know
If you need any further assistance please let us know

closed 2018-10-25 by gtzalik