amzn/amzn-drivers #77: MPLS VPN L3VPN over GRE ping packet size not exceeding 58byes

mohamed-kamar, 2018-06-11:
Hi,
I can't ping with packet size over 58Bytes when I'm using ENA in Amazon instances in MPLS VPN L3VPN over GRE (between two instances) - Example instances types are C5s, however I can ping normally with more than 1500bytes (Amazon internal MTU is 9001) using normal amazon instances drivers (not ENA or VF) - example instances types are T2s, I can reproduce the issue using simple iproute2 commands and one 1 VRF as follows:
              
#Create VRF CUSTC
sudo ip link add CUSTC type vrf table 2012
sudo ip link set dev CUSTC up
sudo ip rule add iif CUSTC table 2012
sudo ip rule add oif CUSTC table 2012
#Create Dummy Interface
sudo ip link add dummy700 type dummy
sudo ip address add 192.168.70.70/32 dev dummy700
sudo ip link set dummy700 up
#Assign the GRE and the Dummy Interface to the VRF
sudo ip link set dummy700 master CUSTC
sudo ip link set gre700 master CUSTC
#Adding MPLS routes
sudo ip rou add vrf CUSTC 192.168.70.71/32 encap mpls 200 dev gre700     
sudo ip -f mpls route add 201 dev CUSTC   

-------------------------------------------------------------------------
AWS VM2:
----------------
#Create VRF CUSTC
sudo ip link add CUSTC type vrf table 2012
sudo ip link set dev CUSTC up
sudo ip rule add iif CUSTC table 2012
sudo ip rule add oif CUSTC table 2012
#Create Dummy Interface
sudo ip link add dummy701 type dummy
sudo ip address add 192.168.70.71/32 dev dummy701
sudo ip link set dummy701 up
#Assign the GRE and the Dummy Interface to the VRF
sudo ip link set dummy701 master CUSTC
sudo ip link set gre700 master CUSTC
#Asdding MPLS routes
sudo ip rou add vrf CUSTC 192.168.70.70/32 encap mpls 201 dev gre700       
sudo ip -f mpls route add 200 dev CUSTC                                  
``` 

Testing:
==========
sudo ip vrf exec CUSTC ping 192.168.70.70 -I 192.168.70.71 -s 300

Note: 
- In the above example I'm using ubuntu with kernel 4.10, but I have also tried 4.14 and 4.16 and all gives the same results 

- We have also upgrdated ENA to 1.5.3g and that's also give the same results 

- "netstat -i" shows the drops in the VRF not the phyiscal or GRE

- dropcatch tools show the looses due to brk_limit function as follows

dropwatch> start
Enabling monitoring...
Kernel monitoring activated.
Issue Ctrl-C to stop monitoring
1 drops at __brk_limit+309db494 (0xffffffffc02c4494)
1 drops at __brk_limit+309db494 (0xffffffffc02c4494)
1 drops at __brk_limit+309db494 (0xffffffffc02c4494)
1 drops at __brk_limit+309db494 (0xffffffffc02c4494)
1 drops at __brk_limit+309db494 (0xffffffffc02c4494)

Thoughts, workaround will be highly appreciated, thanks

akiyano, 2018-06-13:
Hi @mohamed-kamar,

I suggest we take this offline.
Please contact me via email - akiyano@amazon.com

Thanks,
Arthur

mohamed-kamar, 2018-06-14:
Thanks @akiyano , I have sent you some pcap files yesterday.

closed 2020-08-25 by zorikm