amzn/amzn-drivers #78: rpm compilation failure on CentOS 7.5

edtincu, 2018-06-12:
Following the standard instructions (which worked fine until the last CentOS update) I started getting the error below. Based on previous bugs filed it looks like a definition on kcompat.h is now failing.

rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena/kcompat.h:413:6: error: nested redefinition of 'enum pkt_hash_types'
 enum pkt_hash_types {
      ^
compilation terminated due to -Wfatal-errors.

edtincu, 2018-06-12:
Kernel version 3.10.0-862.3.2

rpmbuild/BUILD/ena-1.5.0/obj/default/kernel/linux/ena] Error 2
make: Leaving directory `/usr/src/kernels/3.10.0-862.3.2.el7.x86_64'

snixon, 2018-06-12:
I've been building on Centos 7.5 against the 1.5.3 release after running into the same compilation errors.

akiyano, 2018-06-12:
### Hi @edtincu  @snixon, 

Version 1.5.3 was released to fix the compilation bug you are experiencing.

You can make sure of this by running the following quick test:
1. Find the AMI named "CentOS 7 (x86_64) - with Updates HVM" in your region (for example in N.Virginia it is ami-d5bf2caa) and create an instance with this AMI on an instance type that supports ENA (for example r4.large).
2. Connect to the instance and run the following commands:
```
sudo yum install git
git clone git@github.com:amzn/amzn-drivers.git
cd amzn-drivers/kernel/linux/rpm/ 
sudo yum -y install kernel-{devel,headers}-$(uname -r)
sudo yum -y install gcc make rpm-build
make
rpmbuild --rebuild /home/centos/amzn-drivers/kernel/linux/rpm/ena-1.5.3-1.el7.centos.7.src.rpm
sudo yum -y localinstall /home/centos/rpmbuild/RPMS/x86_64/kmod-ena-1.5.3-1.el7.centos.7.x86_64.rpm
```
3. Reboot the instance. After the instance booted again you should be able to see that you have version 1.5.3 of the ENA driver installed and working. You can see this by running 
`ethtool -i ens3`

@edtincu, I suspect  you tried to compile version 1.5.0 with centos 7.5, because of the folder name  in your error message: "rpmbuild/BUILD/**ena-1.5.0**..."
Please fetch the latest driver version 1.5.3 and use the driver installation instructions again.

edtincu, 2018-06-12:
Thanks, looks like my automation was picking up ena-1.5.0-1.el7.centos.4.src.rpm instead of ena-1.5.3-1.el7.centos.7.src.rpm
I will try now

edtincu, 2018-06-13:
@akiyano Yes, thank you. All working well now

closed 2018-06-13 by edtincu