amzn/amzn-drivers #79: DPDK UDP CheckSum Offload with ENA

Luo-Liang, 2018-06-15:
Hi!

I noticed the UDP checksum offload doesn't seem to calculate correct checksum with DPDK and ENA on EC2.

I have set `udp.dgram_cksum` and `ip.hdr_checksum` both to 0, and have set attribute `PKT_TX_IPV4  | PKT_TX_UDP_CKSUM | PKT_TX_IP_CKSUM;` for the corresponding `rte_mbuf`.

On the receiver side, I'm getting bunch of checksum errors for UDP (but IP seems to be correct):

00:05:23.744861 IP (tos 0x0, ttl 64, id 44761, offset 0, flags [none], proto UDP (17), length 33)
    172.31.2.61.1234 > 172.31.12.72.5678: **[bad udp cksum 0x66e2 -> 0x9975!]** UDP, length 5
        0x0000:  0a5d 098f be18 0ae1 4b05 3dfc 0800 4500  .]......K.=...E.
        0x0010:  0021 aed9 0000 4011 652f ac1f 023d ac1f  .!....@.e/...=..
        0x0020:  0c48 04d2 162e 000d 66e2 504c 494e 4b    .H......f.PLINK
I'm wondering if there is something clearly wrong on my end.

Thanks!

yastreb78, 2018-06-17:
Dear Luo-Liang,

Please double check the following:
1) DEV_TX_OFFLOAD_UDP_CKSUM is configured on device and queue level
2) mbuf.l3_len is set correctly.

If above still doesn't help, we will need more info ( please send to akiyano@amazon.com)
a) which DPDK version is being used
b) Is there way to access your application and reproduce the problem

Luo-Liang, 2018-06-18:
Thanks for the reply! Will report back!

Luo-Liang, 2018-06-19:
Thanks again,
I am fairly new in dpdk and ena, and I need additional clarification.

1. DEV_TX_OFFLOAD_UDP_CKSUM is configured on device and queue level
Would you please elaborate a little bit more on this? I couldn't find any doc as to where to specify this flag.
2. mbuf.l3_len is set correctly.
I did  `buf->l3_len = sizeof(struct ipv4_hdr);`, would this be sufficient?

yastreb78, 2018-06-19:
re: "Configuring offload on device and queue level" - it's about rte_eth_dev_configure() and rte_eth_tx_queue_setup(), please see example in http://git.dpdk.org/dpdk/tree/app/test-pmd/testpmd.c
Eventually, need to make sure that tx offload condition in http://git.dpdk.org/dpdk/tree/drivers/net/ena/ena_ethdev.c#n324 is taken.

Luo-Liang, 2018-06-22:
Thanks! Sorry if I am mistaken, but it doesn't look like there is a `.tx_mode.offloads` in my DPDK version (17.08), do I need to upgrade a new version for this feature?

yastreb78, 2018-06-22:
I was assuming you were using latest DPDK. If you use 17.08 , these steps are not required. Also, it's not required to update to the latest DPDK version to get UDP checksum offload working with ENA.

To continue, it would be helpful if we can get repro of your environment.
If you can, please provide sources , compilation and run instructions to  akiyano@amazon.com and we will continue to analyze it offline.

gtzalik, 2019-03-20:
@Luo-Liang I'm closing this issue for now. please re-open if the issue still exists.

closed 2019-03-20 by gtzalik