amzn/amzn-drivers #80: Programming custom RSS Hash Key and Indirection tables with toeplitz hashing on ENA

nirmalanandj, 2018-06-29:
Hello Experts,

https://github.com/amzn/amzn-drivers/issues/24
Is this limitation still there on ENA on AWS?

We would like to program a RSS Hash key and CPU indirection table entries of our choice with toeplitz hashing through Admin Queue? Is this supported now or it is still a limitation with ENA devices?

What is the max queues RSS queues that can be configured. Is it 128? Has it been benchmarked with queue count greater than 16 or so? Could you please point me to a link?

Regards,
Nirmal

netanelbelgazal, 2018-07-01:
Hi Nirmal,

This feature is still not supported by the ENA device.
It is in our planning and we are going to introduce it in the near future.

The ENA device supports hash indirection table of 128 entries.

ashaffer, 2019-06-24:
@netanelbelgazal Any movement on this issue? Just ran into this trying to get [seastar](https://github.com/scylladb/seastar) to work. It seems there's no way to externally configure the hash function or key when using DPDK. See [my issue](https://github.com/scylladb/seastar/issues/654) on their repository for more details.

akiyano, 2019-07-01:
Hi ashaffer,
Hash key and function are currently not settable.
This feature is planned to be supported in the upcoming future (on some instance types).
I will update this issue once it is implemented.
Sorry for the inconvenience,
Arthur

ashaffer, 2019-07-01:
Can you tell me which hash is being used currently?

sameehj, 2020-01-23:
The default hashing is currently Toeplitz.

Starting from ena driver v2.2.1 the driver supports changing the hash key and hash function as well as the indirection table itself. The support is only for instance types that end with "n", for example C5n instances.

Please note that changing the indirection table is supported on all instance types.

closed 2020-01-23 by sameehj