amzn/amzn-drivers #83: rxclass: Cannot get RX class rule count

SCHEN2015, 2018-09-18:
Description of problem:
We found ENA driver doesn't support getting RX class rule count while executing `ethtool -n eth0` in an AWS instance and this operation will lead an error level message in system log: `ena 0000:00:05.0 eth0: Command parameter 46 is not supported`

Version-Release number of selected component:
ena-1.5.0g
ena-1.5.0K

Maybe we should support this feature some day. Welcome any comments and thank you in advance!

SCHEN2015, 2018-09-19:
Also not support `ethtool -x eth0` which to "get RX flow hash indirection table", but there is no error messages generated in the system log.

BTW, also not support `ethtool -a` (get device pause settings), no error messages in system log.

zorikm, 2018-09-22:
Thanks @SCHEN2015, we are looking into it, and will get back to you with the answers ASAP.

akiyano, 2018-10-10:
Hi @SCHEN2015, 

ethtool -x should work. I tested it with drivers 1.3 - 1.6 and it worked in all of them. Please send me the details of your setup (Region, AMI, instance type) to my email akiyano@amazon.com and I will investigate this.

As for ethtool -a and -n they are indeed not supported and there are no current plans to support them.

Thanks,
Arthur

SCHEN2015, 2018-10-15:
> Hi @SCHEN2015,
> 
> ethtool -x should work. I tested it with drivers 1.3 - 1.6 and it worked in all of them. Please send me the details of your setup (Region, AMI, instance type) to my email [akiyano@amazon.com](mailto:akiyano@amazon.com) and I will investigate this.
> 
> As for ethtool -a and -n they are indeed not supported and there are no current plans to support them.
> 
> Thanks,
> Arthur

Sent. Thanks!

SCHEN2015, 2018-11-06:
Got it, thanks @akiyano !