amzn/amzn-drivers #84: serial8250: too much work for irq4

missinglink, 2018-09-26:
I am investigating unexpected machine reboots and find many of these log lines in journalctl:

```bash
Sep 06 23:47:06 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:06 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:07 ip-172-31-35-48 cloud-init[1314]: [48.0K blob data]
Sep 06 23:47:18 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:18 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:18 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:18 ip-172-31-35-48 kernel: serial8250: too much work for irq4
Sep 06 23:47:18 ip-172-31-35-48 kernel: ena 0000:00:03.0 ens3: Found a Tx that wasn't completed on time, qid 0, index 202.
Sep 06 23:47:18 ip-172-31-35-48 kernel: ena 0000:00:03.0 ens3: Found a Tx that wasn't completed on time, qid 0, index 203.
Sep 06 23:47:18 ip-172-31-35-48 kernel: ena 0000:00:03.0 ens3: Found a Tx that wasn't completed on time, qid 0, index 204.
Sep 06 23:47:18 ip-172-31-35-48 kernel: ena 0000:00:03.0 ens3: Found a Tx that wasn't completed on time, qid 0, index 205.
Sep 06 23:47:18 ip-172-31-35-48 kernel: ena 0000:00:03.0 ens3: Found a Tx that wasn't completed on time, qid 0, index 206.
```

Are these `serial8250: too much work for irq4` messages a cause for concern?
There are many thousands of these loglines leading up until the time the system reboots

akiyano, 2018-09-26:
Hi,

Please send me an instance ID of an instance that is behaving this way to my email: akiyano@amazon.com and we will continue this conversation via email conversation.

Thanks,
Arthur

closed 2018-09-26 by akiyano

bfeinberg43, 2018-12-09:
I am seeing the same error in my ec2 instances, and the hosts will not recover unless they are rebooted.  What was the resolution here?  My hosts are m5 class, using ENA, is this an ENA bug?

akiyano, 2018-12-09:
Hi @bfeinberg43,

A known cause of "serial8250: too much work for irq4" messages, is printing large amounts of output to dmesg. This is not ENA related.

Please send me the instance ID of an instance that is behaving this way, as well as the dmesg when this is happening to my email: akiyano@amazon.com for further investigation.

Regards,
Arthur

nafeabshara, 2018-12-09:
@bfeindberg43

The dmesg on serial8250 is usually an indication that an app is flooding the serial port, which is limited to 115200 baud rate.

Best solution is to reduce verbosity of the appliacation spelling to serial port.