amzn/amzn-drivers #89: ENA Driver 1.6.0 on AMI 2018.3 and  Amazon ECS-Optimized Amazon Linux AMI 2018.03.h

danielvw2003, 2018-11-07:
The installation of the module causes Kernel crash at startup and gives error "Out of Tree Module Taints Kernel." 
Kernel version: 
4.14.72-68.55.amzn1.x86_64

akiyano, 2018-11-08:
Hi danielvw2003,

How did you install the driver? 
looking at https://github.com/amzn/amzn-drivers/issues/88 the RPM installation currently doesn't work on this AMI.

Thanks,
Arthur

danielvw2003, 2018-11-09:
I compiled it using make then replaced the old 1.5 driver and ran dracut to rebuild intramfs

akiyano, 2018-11-12:
The message "Out of Tree Module Taints Kernel" is normal and you would see it no matter how you would install the driver (rpm, DKMS or like you did). It only means that the driver is an out-of-tree module that does not come with the kernel it is built against.

danielvw2003, 2018-11-13:
For some reason though the kernel crashes immediately after loading it.

closed 2018-11-29 by akiyano